gioblu/PJON #239: AnalogSampling - infrared receiver circuit using an infrared (IR) phototransistor

andrei-volkau, 2018-07-31:
**Problem statement:** I am using a couple of pins (separate transmitter and receiver channels). 
`bus.strategy.set_pins(A0,A1);`
It works in case of using 2 IR LEDs. I did that. I want to conduct an experiential with (IR) phototransistor.

**Question:** Could you please share your thoughts about using of infrared (IR) phototransistor in the following way?
[image]

I tried to use the above circuit. I am using `DEFAULT: Default analog reference 5v (5v Arduino boards) `
I am using Arduino Pro Mini (5V, 16 MHz). Unfortunately the above circuit did not work.

gioblu, 2018-08-01:
Ciao @andrei-volkau sorry for the delay in the response. Look here: 
[image]
source (http://www.me.umn.edu/courses/me2011/arduino/technotes/irbeam/irbeam.html)

This schematic shows the emitter always on connected to 5v, what you should change is to connect the positive of the emitter instead of to 5v, to the emitter io pin.

The pull down resistor for the emitter LED is not required, it is required only:
- In single pin/LED setup 
- On the receiver pin for 2 pin setups

gioblu, 2018-08-01:
@andrei-volkau I have never tested AS or PJDLS using phototransistor, it has been tested only using the photodiode (a LED or a photodiode designed for reception only) in positive bias or photo-voltaic mode or more simply  converting light into electricity.

gioblu, 2018-09-17:
Closing for now, feel free to reopen if additional info is required :)

closed 2018-09-17 by gioblu