gioblu/PJON #240: AnalogSampling - read_byte() detects 1s from noise --> can_start() returns False all the time

andrei-volkau, 2018-08-02:
**Case 1**
I was using Arduino Uno in the following way. The IR LED was used for Transmission/Receiving signals. It worked well. 
[image]

**Case 2**
I started to use Arduino Pro Mini in the following way.
[image]

In both cases, we use  1.1v analog reference.

In case 2, `can_start()` function returned False all the time.
`if(read_byte() != 0B00000000) return false;`
`read_byte()` almost never returned 0B00000000

It means that transmission can not be started. 
So the channel is not busy (there is no active transmission of IR radiation), but `can_start()` says that the channel is busy.   

Case 2: Let me show you low_bit, threshold, high_bit, bit_value.
[image]

Usage of solderless breadboard causes noise. So `read_byte()` function tries to read a byte from the noise. The magnitude of the noise is not really high.  The `read_byte()` function detects 1s very frequently. 

Case 1: Let me show you low_bit, threshold, high_bit, bit_value.
[image]

There is no solderless breadboard. So there is no noise. So `read_byte()` function almost never detects 1s. 

How we fixed the issue? We fixed the issue in the following way. 
```
uint8_t read_byte() {
      uint16_t bit_value = 0;
      uint16_t high_bit = 10;
      uint16_t low_bit = 10;
      uint8_t byte_value = 0;
```

We are able to use solderless breadboard based robots after applying the fix now. We are using the latest master. Could you please address the issue in a proper way? We are ready to help. We can test the fix.

andrei-volkau, 2018-08-05:
Hi @gioblu ,
Thank you for the response.

> Having a dedicated pcb would be the best solution to have replicable and reliable results.

We created the following dedicated PCB.
[image]
The PCB is simple. Our first robot includes the following components
1. 2 IR LEDs [each LED is used for Bidirectional Communication];
2. 2 current-limiting SMD resistors  (120Ω)
3. 2 pull down SMD resistors (680kΩ)
4. Arduino Pro Mini 328 - 5V/16MHz
5. Step-Up DC–DC Converter. Here is a picture of it.
[image]
6. one electrolytic filter capacitor, one ceramic filter capacitor

Unfortunately it didn't solved the problem. read_byte() still detects 1s from noise (actually channel is not busy), therefore can_start() returns False all the time. So transmission can not be started. We steel need to hardcode `high_bit` and `low_bit` constants. It helps. So transmission works after that. So the robots are able to communicate with each other. 

Future improvements:
1. We can try to use Arduino Pro Mini 328 - 3.3V/8MHz instead of Arduino Pro Mini 328 - 5V/16MHz. We can use RAW input (MIC5205-3.3BM5, 150mA Low-Noise LDO Regulator). So we can connect  Li polymer accumulator to RAW input. This approach will allow us to get rid from Step-Up DC–DC Converter.  The step-Up DC–DC Converter produces a lot of noise.
2. We can try to add some kind of interference protection to the PCB. I mean that we can protect analog signal from LED on its way to an analog pin.

**I suggest you to use shorter wires to reduce induced interference, if the problem is still there you should just reduce a little the pull down resistance.**
We are using 680kΩ pull down resistor. It seems to be small enough resistance.

**Conclusion**: you suggest to reduce the noise. In case of super low noise,  read_byte() will not dtect 1s from noise when actually channel is not busy.

**Question**: I am not sure that it possible to completely get rid of the noise. So in this case the transmission process will not happen at all.  read_byte() will detect 1s from noise (actually channel is not busy), therefore can_start() will return False all the time.
In case of our best robot, hard coding of the following constants solves the issue. 
```
uint8_t read_byte() {
      uint16_t bit_value = 0;
      uint16_t high_bit = 10;
      uint16_t low_bit = 10;
      uint8_t byte_value = 0;
```
Are you completely sure that the issue is related to hardware only? It means that the only way to resolve the issue is complete reducing of noise. It means that even low level of noise is not acceptable. Is there any software fixes that might address the issue?

gioblu, 2018-08-08:
Ciao @andrei-volkau yes, you are absolutely right, for sure this must be solved in software.
Thank you for your report, looking into it, obviously meanwhile you can get away with your temporary patch. 

My compliments for the great work 👍 
I would be really curious to see a video of the robots roving around!!
Keep me updated and let me know if you need help.

one thing, if you are using continuous rotation servos one interesting thing is to pull a couple of wires from the poles of the dc motor and sample with analogReading the voltage. That is directly proportional to the rotation of the motor.

I have the codebase ready to read that signal and let the robot move receiving cartesian coordinates (although is something I have written 8 years ago could be useful for you now):

====================================

Thanks to this code, you can control the robot with Cartesian coordinates. In the same way, with a simple radio link, in real time obtain the position of the robot. Let's say that the possibilities that are open are almost endless and the game does not stop immediately as with normal robots with continuous rotation servos, which actually before today were rather "limited".

### Integration of Voltages 

I welded two cables to the two poles of the DC motors inside the servomotors. This will allow us to perceive the voltage present in the power supply circuit through the analogue ports. The behavior of this feedback is as follows:

- port1 4.35v gate2 0.25v = full forward
- port1 2.35v porta2 2.35v = stationary motors
- port1 0.25v port2 4.35v = back all

What we need to do now is to frame which pair of 4 cables we have welded will have a high voltage while the robot goes forward. To do this simply call the two commands to move the servomotors (in an empty sketch) and print the 4 values ​​of the 4 soldered feedback in serial mode.

Here is the code you need, **remember to include the Servo.h** library:
```cpp
Servo leftservo; // create servo object to control to servo
Servo rightservo; // create servo object to control to servo
 
void setup () {
 leftservo.attach (9);
 rightservo.attach (10);
 Serial.begin (9600);
}
 
void loop () {
 leftservo.write (180); // the two servos must rotate so that the robot moves forward
 rightservo.write (0);
 Serial.print ("left ENGINE - if high v1Left");
 Serial.print (analogRead (2));
 Serial.print ("if low v2Left");
 Serial.print (analogRead (3));
 Serial.print ("right engine - if high v1Right");
 Serial.print (analogRead (4));
 Serial.print ("if high v2Right");
 Serial.println (analogRead (5));
}
```

As you can see it will be enough to clarify which of the two poles has a high voltage during the forward movement to have found the correct wiring. With this we use the integration "scaling" value weight depending on the cycle duration:
```cpp
movement = v1Right * (endTime - startTime);
```
As you can see the pole with high voltage, during the forward movement it is multiplied by the duration of the loop.

### Wheel movement correction

Knowing the voltage in the two power circuits it is possible to correct the movement of the wheels so as to obtain a straight motion or coordinated wheel rotation. To do this the  `correctStop`, `correctGo` and `correctRotation` are used. Thanks to these three functions the driving of the motors is self-regulated to obtain fluid and coordinated movements.

### The calculation of the distance traveled in mm 
Always using the power supply voltage feedback, let's calculate how many mm the robot has traveled. What we do is integrate the voltages and frame the sum of these in order to obtain a complete and precise revolution of the wheel. To do this, I created a variable:
```cpp
float oneRotationValue = 1125000000; // traveled for a spin of the wheel
```
This variable contains the value of the travelled variable that corresponds to a wheel revolution. To make debugging and setting of this variable just call this command: ** goForward (50, 251); ** where 50 is the speed value between 0 and 100 and 251 are the mm that we ask the robot to travel (in this case our wheels being 8cm diameter have a rolling surface of 251mm). Clearly it is unlikely that the value that I have entered corresponds to you for a turn of the wheel, you will need to calibrate it to obtain a fair amount of precision.

The rest is simple:
```cpp
const float wheelDiameter = 80; // wheel diameter in mm
[... 170 lines cut ...]
//////////////////FUNZIONE XY//////////////////////////////////////////////////
void goXY(float x1, float y1) {
 float dx = x1 - x;
 float dy = y1 - y;
 if(abs(dx) &lt; 2){dx = 0;}
 if(abs(dy) &lt; 2){dy = 0;}
 if(dx == 0 && dy == 0){return;}
 float angle = atan2(dy, dx) * (180 / M_PI);
 rotate(angle);
 goForward(40, sqrt(square(dx) + square(dy)));
}

void loop() {
  goForward(40, 500);
  delay(5000);
  absoluteAngle = 0;
  x = 0; 
  y = 0;
  goXY(200, 0);
  goXY(200, 200);
  goXY(0, 200);
  goXY(0, 0);
  delay(5000);
}
```

gioblu, 2018-08-08:
@andrei-volkau I would iterate the board and try to see if you can read from 4 LEDs and share the estimated position of each robot to the other and keep them in formation using coordinates.

gioblu, 2018-09-17:
Ciao @andrei-volkau problem solved, thank you very much for reporting this.
I am really curious to know how the project is going, feel free to ask for support if necessary.
Closing here, feel free to reopen if required.

closed 2018-09-17 by gioblu