gioblu/PJON #241: Get list of connected devices

drtrigon, 2018-08-08:
I found https://github.com/gioblu/PJON/blob/master/documentation/addressing.md#dynamic-addressing and am wondering whether I can use `bus.check_slaves_presence();` also on a bus with fixed ids? Does it work on LUDP also? A quick try resulted in error like:

> ../../../../../../libraries/PJON/src/PJONMaster.h:62:30: error: expected type-specifier before ‘SoftwareBitBang’
 template<typename Strategy = SoftwareBitBang>

Would automatic listing of device ids for fixed ids be possible?

gioblu, 2018-08-10:
Ciao @drtrigon the listing of available devices with static addressing is a feature that should be added on top and is not provided by the PJON class, primarily not to affect memory footprint and also because should be part of a higher level class. How this should be implemented depends on the limitations of the data-link used or the set of data-links potentially used (if master-slave request-response based or multi-master cyclic broadcast request-response based).

drtrigon, 2018-08-13:
Does that mean the dynamic addressing (or parts of it) can not be (re)used for this?

gioblu, 2018-08-14:
Yes absolutely

drtrigon, 2018-08-17:
I will have a look at this, thanks! My naive first approach would be to use e.g. `PJON_BROADCAST` as mentioned in https://github.com/gioblu/PJON/blob/master/documentation/data-transmission.md and https://github.com/gioblu/PJON/issues/40.

mxbranson, 2019-07-09:
All devices will receive broadcast messages. 
Stick your message processing code in your receiver function and look for packet_info.receiver_id to check if its a zero.
You can also use ports to specify different message types and decode the payload based on the packets port.

closed 2020-06-14 by gioblu