gioblu/PJON #242: Interactive router acting as a switch or router but also as a device

drtrigon, 2018-08-08:
According to https://github.com/gioblu/PJON/blob/master/documentation/routing.md#interactive-router an `Interactive router routes packets as a switch or router but it is also able to act as a device`. So that would mean this router can at the same time have an id and be addressed as device? That would be useful to return values and that way get status and debug info from router/switch/tunnels via bus.

Another question aiming into this direction is, can a single program (on an arduino eg.) have 2 or more (n) different ids and act as multiple devices?

gioblu, 2018-08-10:
Ciao @drtrigon yes absolutely, can be "interactive" and so act also as a device reacting to a packet through its receiver callback. I dont think the class supports multiple ids as you say, but it could be easily implemented on top of the existing classes if required,

drtrigon, 2018-08-13:
How can I assign an ID to a switch?

I think having multiple ids could be very useful - eg. having a very simple temperature sensor device that has 1 value. When having multiple temperatures I would be useful if that would only need to increase the number of devices as no further change to "api" (commands send to device) would be needed.

gioblu, 2018-08-14:
Ciao @drtrigon if I understood correctly, you mean to assign a device id to a certain output present on the same device, instead of relating to different devices? That is a hack, generally device ids are device network addresses.

drtrigon, 2018-08-17:
What I think of is a microprocessor (AVR in my case) that runs code that acts like (a) a switch AND (b) as a device (with own id) at the same time. This is why I also asked for different ids and act as multiple devices.

gioblu, 2018-08-18:
Ciao @drtrigon I am sorry I misunderstood your question, sorry if my answers are not detailed as usual but I am in vacation in a place where wireless and electricity are available as much as water in the sahara :0 I will be back fully available 24 of august.

drtrigon, 2018-08-19:
No problem - enjoy your holidays and answer when you have time again... I will shift to another project during that time!

fredilarsen, 2018-08-24:
@drtrigon The PJONAny constructor takes the device id of that bus as a second parameter. The default device id is PJON_NOT_ASSIGNED, making the switch "invisible" on that bus. You can assign a unique device id to each attached bus.

You cannot assign multiple device ids per physical bus to a switch, but have to add this logic yourself, by checking the receiver id of each incoming packet and replying with send_from_id if you want the device to fake multiple devices.

drtrigon, 2018-09-02:
So I modified the SWBB-TL tunneler like this
```
...
StrategyLink<SoftwareBitBang> link1;
StrategyLink<ThroughLora> link2;

PJONAny bus1(&link1, 99);
PJONAny bus2(&link2);
...
  router.set_error(error_handler);
  router.set_receiver(receiver_function);
  router.set_virtual_bus(0); // Enable virtual bus
  router.begin();
...
```
that makes sense - then when replying as a device on bus1 I had to use
```
bus1.reply(...);
```

It does work. The thing is whenever I got an answer from this device - all other devices do not answer anymore for several seconds. Gives me the impression something wrong/bad is happens. Ideas? Bugs in the code posted?