gioblu/PJON #243: PJON Throughserial on RS485

mrfawy, 2018-08-09:
I'm trying to connect Arduino uno(sender) with pro-mini (atmega328/16MHz/5v) with ThroughSerial over RS485. 

Problem:
I'm getting a lot of Busy/interference  (1667) which affects the throughput. I'm testing with a modified version of network analysis example (modified to work with ThroughSerial) 

I'm using software serial with Baudrate :115200 .. also tried with 9600 with similar results

Here's my current readings: 

Packet Overhead: 10B - Total: 740B
**Bandwidth: 2220B/s
Data throughput: 1480B/s**
Packets sent: 74
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 0
**Busy (Channel is busy or affected by interference): 1667**
Accuracy: 100.00 %

Expected results : lower or 0 value for busy/interference counter .. also serial/RS485 is supposed to have a high throughput .. I checked the Documentation but couldn't find the expected value when using PJON

attempted solutions :
- I tried twisting the A/B wires of the RS485 , also moving them apart but without success
- also tried lowering software serial baud rate to 9600
- tried using UART (hardware) on receiver side 

I intend to use more than 2 devices on the bus, having this much interference with just 2 nodes will be a bigger problem with more nodes I guess .

Here's my  modified  version of the network analysis example 

The sender ( Arduino uno ) 
`
#include <SoftwareSerial.h>
#include <PJON.h>

#define SSerialRX        10  //Serial Receive pin
#define SSerialTX        11  //Serial Transmit pin
#define RS485_pin  12

SoftwareSerial ss(SSerialRX, SSerialTX); // RX, TX

PJON<ThroughSerial> bus(45);

float test;
float mistakes;
int busy;
int fail;
int packet;
char content[] = "01234567890123456789";

void setup() {
  // Initialize LED 13 to be off
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  // Initialize Serial instance used for PJON communication
  Serial.begin(9600);
  ss.begin(115200);

  bus.strategy.set_serial(&ss);
  bus.strategy.set_enable_RS485_pin(RS485_pin);

  bus.begin();

  Serial.println("PJON - Network analysis");
    Serial.println("Starting a 1 second communication test...");
  Serial.println();
}

void loop() {
  long time = millis();
  while(millis() - time < 1000) {

    /* Here send_packet low level function is used to
    be able to catch every single sending result. */

    unsigned int response = bus.send_packet(44, content, 20);
    if(response == PJON_ACK)
      test++;
    if(response == PJON_NAK)
      mistakes++;
    if(response == PJON_BUSY)
      busy++;
    if(response == PJON_FAIL)
      fail++;
  }
  Serial.print("Packet Overhead: ");
  Serial.print(bus.packet_overhead(bus.data[1]) + 1);
  Serial.print("B - Total: ");
  Serial.print((unsigned int)((bus.packet_overhead(bus.data[1]) + 1) * test));
  Serial.println("B");
  Serial.print("Bandwidth: ");
  // length + packet overhead + PJON_ACK
  Serial.print((unsigned int)(test * (20 + bus.packet_overhead(bus.data[1]) + 1)));
  Serial.println("B/s");
  Serial.print("Data throughput: ");
  Serial.print((unsigned int)(test * 20));
  Serial.println("B/s");
  Serial.print("Packets sent: ");
  Serial.println((unsigned int)test);
  Serial.print("Mistakes (error found with CRC): ");
  Serial.println((unsigned int)mistakes);
  Serial.print("Fail (no acknowledge from receiver): ");
  Serial.println(fail);
  Serial.print("Busy (Channel is busy or affected by interference): ");
  Serial.println(busy);
  Serial.print("Accuracy: ");
  Serial.print(100 - (100 / (test / mistakes)));
  Serial.println(" %");
  Serial.println("---------------------");
  // Avoid Serial interference during test flushing
  Serial.flush();

  test = 0;
  mistakes = 0;
  busy = 0;
  fail = 0;
};

`

Code for Receiver 

`
#include <SoftwareSerial.h>
#include <PJON.h>

#define SSerialRX        7  //Serial Receive pin
#define SSerialTX        6  //Serial Transmit pin
#define RS485_pin  8

SoftwareSerial ss(SSerialRX, SSerialTX); // RX, TX
// <Strategy name> bus(selected device id)
PJON<ThroughSerial> bus(44);

void setup() {

  Serial.begin(9600);
  ss.begin(115200);

  bus.strategy.set_serial(&ss);
  bus.strategy.set_enable_RS485_pin(RS485_pin);

  bus.begin();
};

void loop() {
  bus.receive();
};

`

Appreciate your help

mrfawy, 2018-08-09:
Thanks a lot @gioblu .. I tried setting it to 0 and result has much improved 👍 
Packet Overhead: 10B - Total: 2670B
**Bandwidth: 8010B/s
Data throughput: 5340B/s
Packets sent: 267**
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 0
**Busy (Channel is busy or affected by interference): 0**
Accuracy: 100.00 %

I didn't fully understand the last part about multi-master .. documentation shows that single/multi-master is about addressing ,e.g. to allow each node to generate its ID without dependence on a single master .

My use case is that I've several nodes which reads sensor values and can control switches, periodically they send their readings to a node  which is connected to wifi ( it's a bridge to MQTT) .. and when this node receives MQTT message it broadcast the message to all other nodes  so they can operate the switches accordingly .. 

This looks like a single master-slave arrangement I guess .. but in this case can any node initiate the connection to send its sensor reading ? or do you recommend a different approach ? 

Again, thanks a lot for your prompt feedback ..

gioblu, 2018-08-09:
Ciao @mrfawy I would suggest you if the bandwidth is enough to structure the procedure using master-slave approach:

- Master asks for setting sending a request to each slave and waits for an answer
- Each slave answers back its state after receiving the master's request
- Master sometimes broadcasts new state

Doing so you are sure that no collisions can occur, because each slave will transmit only when it was asked providing with a more stable and predictable performance. Probably has no sense to use at all the acknowledgment.

akadlec, 2019-02-14:
Hi there. I am facing same issue here. I have simillar network. One master (NodeMCU) and slaves on Atmega. When slave is connected, search for mater by broadcasting to all. Master will reply. Slave store master address and send few describe packets (tells master who he is IO, etc.) Master store this info.

Now i am facing problem, that after founding master, i am not able to send from slave to master. Getting error `PJON_BUSY` but packet is received by master. I tried setting `TS_TIME_IN` to `0`

I will try to modify communication to master controlled but i need both directions. Eg. slave has buttons and i need to send button click events when they occur.

akadlec, 2019-02-14:
Hmm mabye master controled is the way, it is like modbus right?

gioblu, 2019-02-15:
Ciao @akadlec if you get `PJON_BUSY` trying to send a packet immediately or soon after a packet is received that return value it is normal or to be expected. TSDL to support the synchronous acknowledgment in multi-master mode imposes to all connected devices to wait to transmit and probe the channel for a period longer than the maximum time assigned for an acknowledgement reception, if no data was transmitted in this period, devices can be sure not to disrupt a packet - acknowledgment exchange (or start the transmission in between them) and be sure not to collide with another transmission in already process. See [here](https://github.com/gioblu/PJON/blob/master/src/strategies/ThroughSerial/specification/TSDL-specification-v2.1.md#frame-transmission) 

The simplest solution is to continuously poll the nodes with the master using a request-response mechanism without the use of any acknowledgment and setting `TS_TIME_IN` to `0`. The button press is dispatched as soon as the master rolls back to that node for update. No collision can possibly happen, all should be perfectly stable. If the amount of nodes to be contacted is not enormous and If the latency required in between the button press and the effective actuation is not particularly short it may be the solution.

gioblu, 2019-09-13:
Ciao @akadlec would you share the code?

akadlec, 2019-09-13:
@gioblu it is quite big, but this should be mandatory part:

```
#define PJON_INCLUDE_TS

#define TS_MAX_ATTEMPTS 5

#ifndef PJON_PACKET_MAX_LENGTH
    #define PJON_PACKET_MAX_LENGTH 80
#endif

#define TS_TIME_IN 0

#include <PJON.h>

PJON<ThroughSerial> _gateway_bus(PJON_MASTER_ID);

SoftwareSerial _gateway_serial_bus(NODES_GATEWAY_TX_PIN, NODES_GATEWAY_RX_PIN);

void setup() {
    _gateway_serial_bus.begin(SERIAL_BAUDRATE);

    _gateway_bus.strategy.set_serial(&_gateway_serial_bus);

    _gateway_bus.set_receiver(_gatewayReceiveHandler);
    _gateway_bus.set_error(_gatewayErrorHandler);

    _gateway_bus.begin();
}

void loop() {
    _gatewayReadNodes();

    _gateway_bus.update();
    _gateway_bus.receive(50000);
}

and `_gatewayReadNodes` iterate through all known nodes:

```
void _gatewayReadNodes()
{
    for (uint8_t i = _gateway_reading_node_index; i < NODES_GATEWAY_MAX_NODES; i++) {
        // Composing address, payload and calculating length
        // ....

        uint16_t result = _gateway_bus.send_packet_blocking(
            address,     // Node address
            payload,    // Content
            length       // Content length
        );

    if (result != PJON_ACK) {
        if (result == PJON_BUSY ) {
            DEBUG_MSG(PSTR("[GATEWAY][ERR] Sending packet: %s for node: %d failed, bus is busy\n"), _gatewayPacketName(payload[0]).c_str(), address);

        } else if (result == PJON_FAIL) {
            DEBUG_MSG(PSTR("[GATEWAY][ERR] Sending packet: %s for node: %d failed\n"), _gatewayPacketName(payload[0]).c_str(), address);

        } else {
            DEBUG_MSG(PSTR("[GATEWAY][ERR] Sending packet: %s for node: %d failed, unknonw error\n"), _gatewayPacketName(payload[0]).c_str(), address);
        }

        return false;
    }
    }
}
```

akadlec, 2019-09-13:
and for slave node:

```
#define PJON_INCLUDE_TS

#ifndef PJON_PACKET_MAX_LENGTH
    #define PJON_PACKET_MAX_LENGTH 80
#endif

#include <PJON.h>

PJON<ThroughSerial> _communication_bus(PJON_NOT_ASSIGNED);

SoftwareSerial _communication_serial_bus(COMMUNICATION_BUS_TX_PIN, COMMUNICATION_BUS_RX_PIN);

void setup() {
    _communication_serial_bus.begin(SERIAL_BAUDRATE);

    _communication_bus.strategy.set_serial(&_communication_serial_bus);

    _communication_bus.set_receiver(_communicationReceiverHandler);
    _communication_bus.set_error(_communicationErrorHandler);

    _communication_bus.begin();

    uint8_t node_address = (uint8_t) EEPROM.read(FLASH_ADDRESS_NODE_ADDRESS);

    if (node_address != PJON_NOT_ASSIGNED && node_address > 0 && node_address < 250) {
        _communication_bus.set_id(node_address);
    }
}

void communicationLoop() {
    _communication_bus.update();
    _communication_bus.receive();
}

void _communicationReceiverHandler(
    uint8_t * payload,
    const uint16_t length,
    const PJON_Packet_Info &packetInfo
) {
    uint8_t packet_id = (uint8_t) payload[0];

    uint8_t sender_address = PJON_NOT_ASSIGNED;

    // Get sender address from header
    if (packetInfo.header & PJON_TX_INFO_BIT) {
        sender_address = packetInfo.sender_id;
    }

    switch (packet_id)
    {
        case COMMUNICATION_PACKET_READ_MULTI_DI:
            _communicationReportMultiDigitalRegisters(packet_id, payload, false);
            break;
    }
}

void _communicationReportMultiDigitalRegisters(
    const uint8_t packetId,
    uint8_t * payload,
    const bool output
) {
    // Calculating response, collecting data etc.
    // ...

    uint16_t result = _communication_bus.reply(
        payload,    // Content
        size        // Content lenght
    );

    if (result == PJON_FAIL) {
        #if DEBUG_COMMUNICATION_SUPPORT
            DPRINT(F("[COMMUNICATION] Sending replypacket failed\n"));
        #endif
    }
}
```

gioblu, 2019-09-13:
Ciao @akadlec not bad.
One thing I would do is to set_synchronous_ackonwledge(false); because you don't need an ack for each packet, considering you send a request, and you wait for a response from the slave. This should make the bus quicker removing a lot of overhead. The second thing I would do is to just use send_packet instead of send_packet_blocking you dont need retries, your algorithm already does so.

akadlec, 2019-09-13:
Hi @gioblu thanks for your time, will try it

gioblu, 2019-09-13:
Another thing @akadlec I do see you use SoftwareSerial, why not a hardware serial port, that spare time to the mcu and provide with better performance, give a try also to `ThroushSerialAsync` strategy that has a better performance and can be substituted to TS straight-away.

akadlec, 2019-09-13:
@gioblu right now i can not use HW serial, because testing board is hardwired to other pins, but in the next evolution will give a try. Ok i will try `ThroushSerialAsync` and let you know result :]

akadlec, 2019-09-17:
Hi @gioblu, so i tested `set_synchronous_acknowledge(false);` and found some bugs in my code :) after fixes it is working without any `PJON_BUSY` so i moved to test `ThroushSerialAsync` and yes it was working but i had some `PJON_BUSY` errors so i moved back to `ThroushSerial` with `set_synchronous_acknowledge(false);`

One thing i noticed with `ThroushSerialAsync` is, it look like slave node send response to master request some time 2 times, is it possible? Slave is using `reply`

gioblu, 2019-09-17:
Ciao @akadlec sorry for the late answer if often the response comes twice it means that the transmitter has missed the acknowledgement response. This generally happens because of a too low `TS_RESPONSE_TIME_OUT` or `TSA_RESPONSE_TIME_OUT` value, try to set a higher a value higher than default and see if the problem is solved.

To be sure not to receive twice the same payload, because of a retransmission, you can include the packet id, doing so PJON will deliver to the receiver function the packet only once while filtering duplications. See: https://github.com/gioblu/PJON/blob/master/documentation/configuration.md#packet-identification

akadlec, 2019-09-19:
Hi @gioblu, finaly found issue, was testing in on my breadboards and had some issues with wires. Now is `ThroughSerialAsync` working much better, no `PJON_BUSY` logged, but another one :D `PJON_FAIL` when master is sending packet to the slave. 

Tested packet id, without any success, no packet was transfered.

Tested `TSA_RESPONSE_TIME_OUT` to be `20000` instead of default `10000` and it was worse.

and in master loop i have to use `bus.receive(50000);`

So communication is working, but with some `PJON_FAIL`. Maybe caused by baudrate? or software serial?