gioblu/PJON #244: PJON transmission is not recovering after cable break

gryzli133, 2018-08-12:
In the example:
PJON/examples/ESP8266/Local/SoftwareBitBang/SendAndReceive/ (Arduino MEGA + ESP8266 dev board 12E)

when I disconnect PJON cable for short time the transmission is recovering, but if it takes too long (few seconds) then I will stop the communication. 

The serial monitor results are:

BLINK
BLINK
(...)
BLINK
BLINK
Connection with device ID 44 is lost.

After that the communication stops until I do a restart of the boards. Do I miss a code, where the loop should ends? Or is this some kind of bug?

Thank you in advance for your help!

gryzli133, 2018-08-14:
Yes, I have 160ohm on every input and the bus is pulled down with 100k ohm resitor. When I disconnect the cable for 1-2s there is no message in serial monitor, after reconnect it start to print new BLINK messages. But as soon as I get the message: Connection with device ID 44 is lost.
Nothins more is happening...

gryzli133, 2018-08-14:
Sometimes I don't even get the message: Connection with device ID 44 is lost

Then I have to restart both boards to continue the transmission...

gryzli133, 2018-08-15:
here a short movie what is happening exactly:
https://www.youtube.com/watch?v=k7kuZIPaL98

gioblu, 2018-08-17:
Ciao @gryzli133 sorry for the late answer, I took a 2 weeks vacation in a small island in sicily where internet connection is available as much as water in the sahara. Thank you very very much for taking the time to report so precisely the issue.

This is not a problem, if you look the send and receive example codebase, it is practically a ping pong between your ESP and the Mega. What is happening is that if you interrupt the exchange breaking the connection, the rebound will be broken (as expected) and only a reset will resume the bidirectional exchange :). 

This is a limit of this simple example's codebase rather than a problem of the PJON codebase.

gryzli133, 2018-08-17:
Thank you! Now is everything clear. I'm not a good C programmer, because I used to program only automation PLCs professionally. But if it comes to use and combine some libraries to work on one project, then want to be sure each of the lib is working 100%. 

I'm working on some smart home projects, I have also shared few codes already, that I think most of Poland is using now for control Lights and Roller Shutter (in Smart Home DIY projects). These are based on MySensors and are tested with Domoticz to work together in Smart Homes. Now I'm planning to build an Intelligent Floor Heating room thermostat System. Due to no standard of cabling most of people have between 3 and 8 wires for that purpose in each room. I don't want to have anything wireless, this is why I though I will make an Arduino Mega as a Gateway for Domoticz (using MySensors), that one will be placed centrally and it will control the valves (and the pump/heating unit) for floor heating using some smart algorithm. This will be based on setpoints and actual temperatures that comes from each room. The room thermostat will be based on BME280 sensor, OLED display and ESP8266 (without WiFi). The communication in best way would be by PJON. This would enable to use that system even with only 3 wires (+, -, signal). Now the hardest is to combine OLED, BME and PJON to work parallely. OLED and BME are on i2c bus that is working software based due to leak of HW i2c on ESP. Do you have any idea how to apply PJON to work reliable between Arduino Mega and multiple ESP? And it should not disturb the i2c bus at the same time.

I know that someone was implementing MySensors transport layer using PJON, but it is bit outdated. It would be a good solution to do it that way, but my programming skills are to weak to rework that PJON transport layer in less than one year ;-) 

Would you like to pull request and support officially MySensors? I also have no idea how it works on github... 

I think this one is a very interesing project, but I need some support to bring it to life...

Anyway many many thanks for PJON! :-)

Best Regards from Bodensee

gioblu, 2018-08-28:
Ciao @gryzli133 I will dedicate some time on the MySensors transport layer.
Have you looked at https://github.com/fredilarsen/ModuleInterface by @fredilarsen ?

gryzli133, 2018-09-06:
Thank you for the link, but I would like to stay on common systems like MySensors and Domoticz/Home Assistant. The only disadvantage in MySensors is no reliable wired communication, but if prepare PJON for MySensors then I will have all I need ;-) Thank you in advance!

4ib3r, 2018-11-12:
Hi, I implemented PJON based transport for MySensors, at this moment this exist in my fork: https://github.com/4ib3r/MySensors. I wait for pull-request: #263 and then i will create pull-request for MySensors.

gioblu, 2018-12-05:
Ciao @gryzli133 how its going? See here: https://github.com/mysensors/MySensors/pull/1230

gioblu, 2019-01-08:
Ciao @4ib3r happy new year and thank you for your support. I saw that sadly your pull request was closed due to inactivity. Can I in any way help you to let the pull go through, a lot of people contacts me asking for the MySensors support, its sad we were almost there but it got stuck :).

Thank you again.

4ib3r, 2019-01-08:
Hi, thank you for wishes.

Unfortunately, at now I don't have time to clean and update code.

gioblu, 2019-07-12:
Ciao @gryzli133 @4ib3r I beg you give a try to https://github.com/mysensors/MySensors/tree/b24711fdbe2e482867574abc44dc258bd3884e60

It is the pull request containing the PJON transport for MySensors, the developers of MySensors need some feedback from experienced users, if you can give it a try and let them know how it goes here:
https://github.com/mysensors/MySensors/pull/1278

closed 2020-12-22 by gryzli133