gioblu/PJON #247: PJON Sniffer (SWBB, LUDP, ...)

drtrigon, 2018-09-02:
As noted in https://github.com/gioblu/PJON/issues/222#issuecomment-411489899 I was able to build a sniffer for the ThroughLoRa strategy quite easily. What about other strategies? It would be very useful to have some kind of monitor that is able to watch what actually happens on the bus for debugging. I do understand that for the different strategies and media involved completely different sniffer would be needed. As noted I can provide one for the TL strategy and here is what I think would be needed for others:

* SWBB, AS; needs a logic analyzer based approach
* LUDP, GUDP, ETCP; need a network sniffer based approach (e.g. wireshark or python script)
* TS; USB port sniffer (wireshark) or logic analyzer

gioblu, 2018-09-02:
Ciao @drtrigon thank you very much, yes the LoRa sniffer is for sure an interesting contribution.
I agree with you about SWBB, AS and OS they need an oscilloscope or a logic analyzer depending on what has to be observed.  @drtrigon can I ask you to write me to my email (so you will not epose yours: gioscarab@gmail.com) I would like to show you one thing.

jcallano, 2020-06-12:
any decoder available for pulseview for PJDL SINGLE WIRE BUS?

gioblu, 2020-06-14:
Ciao @jcallano I don't think they are available already. I for now still use old-style oscilloscope.

jcallano, 2020-06-16:
I will try to write a PJDL decoder for sigrok - pulseview open source logic analyzer software. Do you have a flow chart of the process of decode used in the library to use as start point? Sigrok decoder wiki page at https://sigrok.org/wiki/Protocol_decoder:Pjon,

gioblu, 2020-06-16:
Ciao @jcallano thank you for your support :)

I suggest you to check out the specification of PJDL:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v4.1.md

Probably the clearer description is the code itself:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/SoftwareBitBang.h

If you need any help or direct support join our gitter chat: https://gitter.im/gioblu/PJON