gioblu/PJON #248: Error on PJONMaster.h example with ESP8266 or ESP32

fermuch, 2018-09-02:
Hello!

After trying the `MasterSlaveAutoAddressing` example, I found the same error on both ESP8266 and ESP32 chips.
This error happens both with v11.1 and latest master as of this day.

Stack trace:
```
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
⸮
Soft WDT reset

ctx: cont 
sp: 3ffffd20 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  b26278b4 4c957f2d 3ffee9bc 00000014  
3ffffef0:  b26278b4 a4a36a7b 4c957f2d 5851f42d  
3fffff00:  feefeffe feefeffe 4020740e 3ffefc14  
3fffff10:  40106530 00000000 0000000f 40106561  
3fffff20:  40106530 3ffee8bc 3ffee8bc 402029cc  
3fffff30:  00000000 3ffee8bc 00000032 40203510  
3fffff40:  00000000 00000000 3ffee8bc 40202e1b  
3fffff50:  3ffee8da 3fffff90 3ffee8be 3ffeeeec  
3fffff60:  3ffee8bc 00a26c81 0003d090 40203804  
3fffff70:  007e5162 3ffee8c7 3ffee8bc 4020388c  
3fffff80:  00000037 00000000 00000001 3fffefb0  
3fffff90:  402014cc 0000038f 0000038f 402014e7  
3fffffa0:  3fffdad0 000000fe 3ffee8bc 40203930  
3fffffb0:  feefeffe 00000000 3ffeeee5 40205bd8  
3fffffc0:  feefeffe feefeffe 3ffe85dc 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

Parsed stack trace:

[image]

gioblu, 2018-09-02:
Ciao @fermuch which toolchain you are using?
Thank you very much for your report.

gioblu, 2018-09-02:
Ciao @fermuch, sorry, we have still to work out compatibility for the `PJONMaster` and `PJONSlave` classes for non-avr machines. The problem may be caused by the classes procedure that it is for now completely blocking (that is generally not appreciated by ESP8266/ESP32.

fermuch, 2018-09-02:
Oh that's a real bummer. What would you recommend to have auto assigned ids in a mixed attiny and esp environment?

gioblu, 2018-09-03:
Ciao @fermuch I am sorry, ATtiny does not have much memory available and I suspect `PJONSlave` class would consume most of the available progmem and ram. I think, for now, a custom layer on top of PJON should be evaluated.

maniekq, 2018-10-25:
I have successfully run MasterSlaveAutoAddressing example between two ESP32s. However, when trying to run it between ESP32 (as master) and Pro Mini (as slave), slave cannot acquire id. Slave is correctly receiving message "Master says hi!".
@gioblu Do I understand correctly that it is currently expected behaviour because of the "compatibility for the PJONMaster and PJONSlave classes for non-avr machines" you've mentioned?

gioblu, 2018-10-26:
Ciao @maniekq a couple of questions, is the Pro Mini 8 or 16MHz? 
Are you using pin 12 or 25 on ESP32 (the only one supported for now)?

maniekq, 2018-11-05:
Hi @gioblu. Sorry for delayed answer. 
I am running PJONMaster on DOIT Esp32 DevKit v1 on pin 25 (tried on pin 12 as well, but the result is the same). I've tried with both level shifter and direct wiring - results are the same.
I am using PlatformIO instead of Arduino IDE. 

Output from master (ESP32):
```
List of slaves known by master:

List of slaves known by master:

List of slaves known by master:
[...]
```

Output from slave (Pro Mini):
```
PJONSlave error: master-slave id request failed.
Received: M a s t e r   s a y s   h i !
Received: M a s t e r   s a y s   h i !
Received: M a s t e r   s a y s   h i !
Received: M a s t e r   s a y s   h i !
Received: M a s t e r   s a y s   h i !
[...]
```

maniekq, 2018-11-05:
I've started to run NetworkAnalysis tests again to add output and I found out that it doesn't work any more on Pro Mini port 11. I switched to port 12 and then everything including MasterSlaveAutoAddressing example started to work. 
Now I got an impression that it was working with port 11, because I've moved to it from port 10 which I had problems with (which I've described earlier in #260). Either I am destroying ports of my Pro Mini one by one or just have messed up with the port numbers. I will continue testing for some time to find it out.

Anyway, PJON MasterSlaveAutoAddressing is working fine between ESP32 and Pro Mini right now so from PJON implementation perspective there is no issue. 
Thank you @gioblu again for your help. I think this issue can be closed now.

gioblu, 2018-11-06:
Ciao @maniekq to be sure please include the current limiting resistors as described in the SoftwareBitBang README and PJDL specification. Thank you for reporting back results.

Happy tinkering :)

closed 2018-11-06 by gioblu