gioblu/PJON #250: PJON over ESP-NOW ?

xlfe, 2018-09-09:
Hi all

I have a partially working implementation of PJON over ESP-NOW (running on ESP32) 

https://github.com/xlfe/PJON/blob/feature/espnow-32/src/interfaces/ESP32/ESPNOWHelper.h
https://github.com/xlfe/PJON/tree/feature/espnow-32/src/strategies/ESPNOW

It's a work in progress. Currently I can get two ESP32s to send approx. 20 small messages (~5 bytes) per second but I get a HEAP corruption error after running for a couple of minutes.

So definately needs more work to track down that bug.... 

If anyone's interested please take a look. 

Once I have ironed out the bugs @gioblu I'd be very happy to submit a pull request for the support to be merged.

One of the cool things (that's working) is that you don't need to bus.strategy.add_node(id, mac_address) - auto-registration (thanks @fredilarsen ) is supported through broadcast packets and then it automatically moves to unicast once the id/mac combo is registered.

Todo : 
* add doco
* add encryption keys and 802.11 channel to PJON setup (so they dont have to be specified in the esp-idf sdkconfig
* find and fix the memory bug!
* better thread handling

for information at this stage

gioblu, 2018-09-10:
Ciao @xlfe cool stuff :) for sure a lot of users will be empowered with this. 
Feel free to ask for support if that is required, if I can help I am here.

xlfe, 2018-09-14:
I've fixed the bugs and cleaned up the code a fair amount and it's working well now with no crashes.

I added an example https://github.com/xlfe/PJON/tree/feature/espnow-32/examples/ESP32/ESPNOW

If anyone wants to test it that would be great.

gioblu, 2018-09-14:
Ciao @xlfe I am reading through the code, simple and effective.  My compliments, with good antennas this strategy could enable a lot of use cases where decentralization is the key requirement. It is now late here but tomorrow I will for sure run your code and see what happens, thank you for your support and passion @xlfe .

fredilarsen, 2018-09-14:
@xlfe My compliments for adding what seems to be a great protocol choice for the ESP32.

Your code seems clean and efficient. A few mallocs in there that I guess were the cause of your initial heap problem(?), but they seem fully protected now.

The strategy of going from broadcast to direct connection I have used before myself in other contexts, but your implementation gave me some inspiration to maybe improve the GlobalUDP strategy some more. It already does autoregistration of incoming packets so it can reply to them, but it could actually do an initial plus scheduled broadcast to avoid all manual registration of nodes inside the local network segment. That would give a strategy that is not limited to the local network but would require registration only of remote nodes. Is it OK for you if I get inspired by you to do this?

A few questions:
1. Can ESP8266 support be added to your ESPNOW strategy -- it seems ESP8266 also supports ESP-NOW?
2. Do you think running the ESPNOW strategy will block GlobalUDP, LocalUDP and EthernetTCP strategies from being used at the same time in a switch or router? (A use case for this is to allow a master running on a RPI or PC to talk to ESP-NOW devices thorugh the switch).

gioblu, 2018-09-14:
Ciao @xlfe I have studied more in detail the helper, I see a couple of things that may be improved in the encoding/framing to reduce the chances of passing up to the network layer false positives:
- A safer way to separate frames can be, as you do, add a magic header and the length of the frame, but also a footer, that signals the end of the frame. You can see this implemented in ETCP.
- Check the equality of the length encoded in the frame and the length of bytes effectively received until the footer occurred. That would add to ESP-NOW a basic consistency check (also this is implemented in ETCP, instead in TS you can see the same approach in a blocking scenario not including the length at the strategy layer but using the detected length of the network layer passed down to the `receive_string` function.

I agree with @fredilarsen that supporting also ESP8266 would be great, I can make some experiments with this if can help. 

Thank you very much again for your support 👍

xlfe, 2018-09-14:
@fredilarsen and @gioblu - glad you like the addition! 

Q1 - esp8266 support. Yes this should be possible - I started with ESP32 because I have been looking at these more recently, but it should definitely be possible to support esp8266 with a bit more work.

Q2 - Blocking. Very good question. At the moment sending blocks until the MAC layer callback that says it has received the data. The obvious use case is to use a esp32 as a bridge between ESPNOW and something else, eg GlobalUDP or TS or TSA. I plan to try this, but haven't got there yet. I think the design of the ESP32 means the blocking shouldn't affect the Wifi thread (or serial) but it depends how it's configured and which CPU core runs which task. In practice it might not matter- testing should confirm this I suppose.

@gioblu - the encoding/framing could definitely be improved - I'll take a look at ETCP and TS. One limitation of ESPNOW is that packets are limited to 250 bytes - so with overhead PJON packet must be less than that.

Thanks for the feedback!

xlfe, 2018-09-14:
> Your code seems clean and efficient. A few mallocs in there that I guess were the cause of your initial heap problem(?), but they seem fully protected now.

The problems I had at the start were caused by multiple things - using a variable across threads (maybe) and a threading model that was more complicated than need be. Simplifying meant I was more confident that all malloc/frees we're handled correctly, etc. I also discovered components in esp-idf and the main code use different #defines which caused another problem I had to use JTAG to figure out what I was doing wrong.

gioblu, 2018-09-14:
Ciao @xlfe I understand the need to contain the overhead, probably adding 1 byte as end frame delimiter could enable the use of the length as a consistency check.

gioblu, 2018-09-14:
Sorry @xlfe I sometimes may seem hit by over cautiousness, PJON specifies a network layer supporting 4 billion busies and more than 1 trillion devices. In the rare case where this could be applied at its own full potential I would not want any of us to be the cause of someone else's problem. I think it is better to be mathematically sure to cover the operational envelope described in the specification.

I am not good in calculating probabilities, but I think a 4 bytes header and most of the length byte values could occur in pseudo-random data in the order of once in few billions attempted packet receptions. That would mean, during each attempted reception of around one device for each bus, at least one would receive a false positive that would be pushed up to PJON. If a false positive comes from the data-link is not the sort of error that the crc it is made to detect, because the received crc is pseudo-random data. There are chances (higher using crc8) that a false positive could pass through and effectively be parsed by an application. Each attempted reception of all trillion devices would cause in average 200 of them to receive a false positive. A false positive could breach not just one device but as many are attempting reception that are in range, if that is not directed, if a router is breached could wrongly route packets to random ids part of the false positive content to devices very far in the network with unpredictable results.

Adding a 1 byte end delimiter so a 4 bytes header + 1 length + 1 end delimiter encoding, without taking in consideration the effect of the length consistency check or the chances to effectively receive the correct amount of data, so counting raw possible states, a false positive would occur once in 2 ^ (8 bit * 6 bytes) 281.5 trillions attempted receptions that to me seems safer. 

What do you think?

gioblu, 2018-09-14:
I suggest you to take a look to [this paper](https://users.ece.cmu.edu/~koopman/thesis/etran.pdf) that describes the length corruption vulnerability of CAN, or to rely on a broken cyclic redundancy check to discard corrupted frames, that is what would occur in the rare cases where a random acceptable header and length are wrongly found, PJON v3.0 resolves that at the network layer with the header crc immediately after the length.

>  Although this probability is small, the number will be magnified by the size of the vehicle fleet.

xlfe, 2018-09-15:
@gioblu thanks for the suggestion. 

I've changed the magic so it has 2 bytes at the start of the packet and 2 bytes at the end, and I have also added the length as an XOR on the magic, so now each packet is checked for magic, length and start and end integrity

https://github.com/xlfe/PJON/commit/ff1a6ef1481ed0cdac3b87b1a63318df5e34fbfd

Let me know what you think :D

gioblu, 2018-09-15:
Ciao @xlfe thank you very much and sorry if I was prolix :)
So for a 2 bytes magic header a false positive should occur once in 2^24 (2 bytes magic header + length) or 16777216 attempted packet receptions of pseudo-random data (not considering the nice and for sure effective length xor based consistency check you made). Maybe it could have sense to reduce the chances the parsing procedure starts for a false positive that then is determined erroneous thanks to the end delimiter, to also reduce the time lost to try to parse invalid data. In this case If you want to get away with 5 bytes of overhead you could use a 3 bytes magic header, length, nice XOR length verification you made and a single byte end delimiter. All this to trigger erroneous initial parsing once in 2 ^ 32 (3 byte magic header + length) or 4294967296 attempted receptions of pseudo-random data still not considering length consistency check. All in all with 5 bytes should be in the order of once few thousands of trillions, probably more than enough :)

gioblu, 2018-09-15:
@fredilarsen what do you think?

fredilarsen, 2018-09-15:
@gioblu, @xlfe This is a purely packet based protocol like UDP, if I have understood correctly. Even in unicast mode. So it all boils down to recognition and verification and not packet separation like with TCP or serial.

To minimize the time spent on a potential "alien" packet I guess a long (4 byte?) magic header plus the receiver id at the start would allow the packet to be left unprocessed and discarded quickly if not a PJON packet or not meant for this device. 
Everything else than the payload could be dropped, I guess, as the total packet length is available and the payload being a PJON packet would have to pass the PJON packet integrity check including CRC.

fredilarsen, 2018-09-15:
With unicast, only the receiver will pick up the packets meant for it, meaning that we could drop my suggestion of filtering on the PJON id, leading back to the original implementation of only a four byte header :-)

Sorry @xlfe for suggesting complications when your original implementation probably was the best way... at least I got to think it through :-)

xlfe, 2018-09-16:
@gioblu @fredilarsen don't apologise! I think it's great that the code is examined thoughtfully. No problem at all :) 

Happy to take constructive criticism always. (I might not always be able to implement the suggestions, but I figure that's okay because open source means anyone can contribute :D)

My plan is - eventually - to use it in a home automation system - I'm working on the other bits of it so its a work in progress - for examples - https://github.com/xlfe/reticul8

gioblu, 2018-09-16:
@xlfe I am sorry for being hit by over cautiousness :)

gioblu, 2018-09-20:
Ciao @xlfe would you send a pull request to record your contribution? 
I think many users will make use of your creation.

Thank you again for your support.

gioblu, 2018-09-27:
Ciao @xlfe I am studying your contribution, one question: What should be changed to let a more entry level user to run the examples using just the Arduino IDE and the core downloaded from the board manager?

I have tried to compile one of the examples but I get those errors:
```cpp

In file included from C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/strategies/ESPNOW/ESPNOW.h:24:0,

                 from C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/strategies/PJON_Strategies.h:54,

                 from C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/PJON.h:96,

                 from C:\Users\gioblu\AppData\Local\Temp\arduino_modified_sketch_962660\Receiver.ino:9:

C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/interfaces/ARDUINO/ESPNOWHelper.h:32:1: error: 'wifi_country_t' has no non-static data member named 'max_tx_power'

 };

 ^

C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/interfaces/ARDUINO/ESPNOWHelper.h: In member function 'bool ENHelper::begin(uint8_t, uint8_t*)':

C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/interfaces/ARDUINO/ESPNOWHelper.h:154:53: error: 'ESP_ERR_NVS_NEW_VERSION_FOUND' was not declared in this scope

       if(ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {

                                                     ^

C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/interfaces/ARDUINO/ESPNOWHelper.h: At global scope:

C:\Users\gioblu\Documents\Arduino software\arduino-1.8.5\libraries\PJON\src/interfaces/ARDUINO/ESPNOWHelper.h:34:20: warning: 'TAG' defined but not used [-Wunused-variable]

 static const char *TAG = "espnow";

                    ^

exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.
```

Looking at the wifi_country_t I see it is hardcoded here:
https://github.com/gioblu/PJON/blob/master/src/interfaces/ARDUINO/ESPNOWHelper.h#L26

gioblu, 2018-09-27:
@xlfe to be able to compile an example using the ESPNOW strategy I needed to:
- Comment `max_tx_power: 80, // Level0`
- Remove `|| ret == ESP_ERR_NVS_NEW_VERSION_FOUND`

gioblu, 2018-09-27:
@xlfe testing with 2 ESP32 Heltech WiFi LoRa 

I get from one side:
```
PING/s: 44.96
PING/s: 45.95
PING/s: 43.96
PING/s: 45.91
PING/s: 44.96
PING/s: 44.96
```

from the other:
```
PONG/s: 44.96
PONG/s: 44.96
PONG/s: 44.96
PONG/s: 44.96
PONG/s: 44.96
PONG/s: 43.96
PONG/s: 44.91
PONG/s: 44.91
PONG/s: 45.95
PONG/s: 44.96
```

gioblu, 2018-09-27:
@xlfe could have sense to have something like this in the helper:
```cpp
#ifndef PJON_ESPNOW_COUNTRY
  #define PJON_ESPNOW_COUNTRY "AU"
#endif

#ifndef PJON_ESPNOW_POLICY
  #ifndef PJON_ESPNOW_COUNTRY
    #define PJON_ESPNOW_POLICY WIFI_COUNTRY_POLICY_AUTO
  #else
    #define PJON_ESPNOW_POLICY WIFI_COUNTRY_POLICY_MANUAL
  #endif
#endif

#ifndef PJON_ESPNOW_SCHAN
  #define PJON_ESPNOW_SCHAN 1
#endif

#ifndef PJON_ESPNOW_NCHAN
  #define PJON_ESPNOW_NCHAN 14
#endif

static wifi_country_t wifi_country = {
  cc:     PJON_ESPNOW_COUNTRY,
  schan:  PJON_ESPNOW_SCHAN,
  nchan:  PJON_ESPNOW_NCHAN,
  policy: PJON_ESPNOW_POLICY,
  // max_tx_power: 80 // Level0
  //error: wifi_country_t has no non-static data member named 'max_tx_power'
};
```

For then let the user do this in the sketch:
```cpp

#define PJON_ESPNOW_COUNTRY "IT"
#include<PJON.h>

...

```
?

Thank you very much for your support and compliments for ESPNOW 👍

xlfe, 2018-09-29:
@gioblu that all looks reasonable to me! Glad to see it working with the Arduino IDE too :D

gioblu, 2018-09-29:
Ciao @xlfe thank you for your support, I will add the changes suggested here, the arduino IDE compatible examples and minor changes to codestyle in the helper.

gioblu, 2018-09-29:
- WiFi country config constants: https://github.com/gioblu/PJON/commit/e3a6ade96cce20294d8d26faefe2f6705a39064c
- Avoid Arduino IDE errors / codestyle: https://github.com/gioblu/PJON/commit/9ceb5af8b89e76fa52a163e6521677ca4d1d8e6e
- Arduino IDE example and esp32-idf example moved in its own dir: https://github.com/gioblu/PJON/commit/6e8b7bd33cd5b7db150fa3992b45d0c7b736cb46

I have tested the changes and all works fine using Arduino IDE and I think all the rest should be unchanged except the commented lines who were causing compilation errors, I am not sure if the dir change of the idf examples can cause problems because I still have not tested in this configuration.
Can you give it a try?

Thank you very much :)

gioblu, 2018-09-29:
Ciao @xlfe I have modified the PingPongArduinoIDE to run the exchange at the maximum speed possible, I obtain this result:
```cpp
PONG/s: 175.82
PONG/s: 171.83
PONG/s: 168.66
PONG/s: 164.51
PONG/s: 167.83
PONG/s: 173.83
PONG/s: 171.66
PONG/s: 172.83
PONG/s: 164.84
PONG/s: 177.82
PONG/s: 171.66
PONG/s: 167.83
PONG/s: 175.82
PONG/s: 159.68
```

Not bad! see: https://github.com/gioblu/PJON/commit/db42046c207b689b01c84c235f0bee2fe56bfeb5

gioblu, 2018-10-09:
Ciao @xlfe how its going? I have started practical testing of all architectures and strategies, preparing for release, long job. Have you tried the esp32-idf example?

Thank you very much for your support.

xlfe, 2018-10-10:
Hi @gioblu sorry I got distracted working on https://github.com/xlfe/reticul8/ !

I've just tried compiling your changed code using esp-idf and I get the follow errors:-

```C
n file included from PJON/src/strategies/ESPNOW/ESPNOW.h:24:0,
                 from PJON/src/strategies/PJON_Strategies.h:54,
                 from PJON/src/PJON.h:96,
                 from main.cpp:10:
PJON/src/interfaces/ARDUINO/ESPNOWHelper.h: At global scope:
PJON/src/interfaces/ARDUINO/ESPNOWHelper.h:53:1: sorry, unimplemented: non-trivial designated initializers not supported
 };
 ^
PJON/src/interfaces/ARDUINO/ESPNOWHelper.h:53:1: warning: missing initializer for member 'wifi_country_t::policy' [-Wmissing-field-initializers]
```

The following diff seems to fix it

```diff
   cc:     PJON_ESPNOW_COUNTRY,
   schan:  PJON_ESPNOW_SCHAN,
   nchan:  PJON_ESPNOW_NCHAN,
+  max_tx_power: 80,
   policy: PJON_ESPNOW_POLICY,
-  // max_tx_power: 80 // Level0
-  //error: wifi_country_t has no non-static data member named 'max_tx_power'
 };

 static const char *TAG = "espnow";
```

And it compiles without problem. I think the key thing is (for the esp-idf compiler) that all items are initialised in the order they are defined in the original struct (hence swapping max_tx_power and policy) and all items in the struct must also be initialised (hence including policy in the struct).

Does that cause any problems with the Arduino IDE?

gioblu, 2018-10-10:
Ciao @xlfe thank you for testing it, on Arduino IDE I get:
```
error: wifi_country_t has no non-static data member named 'max_tx_power'
```
if `max_tx_power: 80` is present in the struct, I am not completely sure how I can use the preprocessor to conditionally define that depending on the toolchain used. Do you have any suggestion?

xlfe, 2018-10-10:
Ah - [perhaps you can use](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#preprocessor-definitions) 

```C
#ifdef ESP_PLATFORM
```

to detect if the build happens with esp-idf?

gioblu, 2018-10-13:
@xlfe could be a possibility to let the user define in the program the `wifi_country_t` struct as required depending on the toolchain? Could we define in the example the `wifi_country` struct, in that case the configuration constants I have created would not be required anymore and we could avoid the problem defining it differently in the 2 examples.

xlfe, 2018-10-13:
> @xlfe could be a possibility to let the user define in the program the `wifi_country_t` struct as required depending on the toolchain? Could we define in the example the `wifi_country` struct? in [...]

@gioblu yeah that would work I think

gioblu, 2018-10-14:
Ciao @xlfe thank you very much for your feedback, I will make the change and relate it to this issue soon :)

gioblu, 2018-10-20:
Ciao @xlfe I have pushed the changes, thank you for your feedback. The Arduino examples are running fine. Can you give a try to the esp-idf example?

xlfe, 2018-10-20:
@gioblu tried the changes last night - all working well on esp-idf!

Thanks

closed 2018-10-20 by xlfe

gioblu, 2018-10-20:
Thank you @xlfe for this beautiful and useful strategy.