gioblu/PJON #258: check logic for packets sent to router _device_id where _router = true and synchronous_ack=true

xlfe, 2018-10-10:
I'm just trying out using a PJON<Any> bus as a router by doing bus.set_router(true); and I've found that once I enable the router functionality latency is increased by approx 10 times.

My setup is similar to : 

```C
StrategyLink <ThroughSerialAsync> *link_tsa = new StrategyLink<ThroughSerialAsync>;
link_tsa->strategy.set_serial(&Serial);
bus = new PJON<Any>(1);
bus->strategy.set_link(link_tsa);
bus->set_asynchronous_acknowledge(false);
bus->set_synchronous_acknowledge(true);
bus->set_router(true);
bus->begin();
```

If I make the following changes in [PJON.h](https://github.com/gioblu/PJON/blob/master/src/PJON.h#L478)

```diff
      if(data[1] & PJON_ACK_REQ_BIT && data[0] != PJON_BROADCAST)
-        if((_mode != PJON_SIMPLEX) && !_router)
+        if(_mode != PJON_SIMPLEX)
          strategy.send_response(PJON_ACK);
```

Then the latency of communicating with the devices goes back to expected. 

My guess is that this part of receive() in PJON.h assumes all packets received by the device are to be handled by some custom routing logic (where the ACK is sent back) - but wouldn't it be correct for the bus to handle packets sent to _device_id as per normal, even if _router = true?

gioblu, 2018-10-10:
Ciao @xlfe thank you for raising this issue.
With the change you propose also a packet which recipient's bus id is different from the one of the router device would trigger an ack response, that would be wrong / misleading.

What would be required to do is to check again if both device id and the bus id are the same as the one contained in the packet, or save this info in a bool earlier while parsing the packet and check it afterwards there (that would probably be a faster solution at the cost of few bytes of program overhead in the implementation). What do you think?

xlfe, 2018-10-10:
> What would be required to do is to check again if both device id and the bus id are the same as the one contained in the packet, or save this info in a bool earlier while parsing the packet and [...]

I think that makes sense - should I submit a pull request?

gioblu, 2018-10-12:
Ciao @xlfe  sorry for the delay.
Thank you very much for the PR I will now check compilation on Arduino IDE.

About the router issue:
>With the change you propose also a packet which recipient's bus id is different from the one of the router device would trigger an ack response, that would be wrong / misleading.

With that i meant is that here:
https://github.com/gioblu/PJON/blob/master/src/PJON.h#L455

The bus id filtering does not occur in the case `_router` is set true. So if a packet with a different recipient's bus id is picked up by the device in router mode your suggested device id check afterwards would let a sync ack to be sent when it is not supposed to do so.

Although I am not sure this is the natural behavior is expected from a switch/router, and I would discuss further this potential change (@fredilarsen what do you think about this?). I am also not sure if have sense to make the whole stack heavier to spare speed in this particular case at the additional cost of potential misleading behavior relaying to the receiver function packets that are already acked and should be handled to avoid ack duplication.

If has sense to be implemented, a solution may be to add a bool to save if the `router_bus_id` is effectively equal to the one that is being received here: https://github.com/gioblu/PJON/blob/master/src/PJON.h#L397
```cpp
bool router_bus_id = true;
```
Avoid the _router check here: https://github.com/gioblu/PJON/blob/master/src/PJON.h#L455

Change this line where to set the state false if mismatch occurs:
https://github.com/gioblu/PJON/blob/master/src/PJON.h#L460
```cpp
if(!_router) return PJON_BUSY;
else router_bus_id = false;
```
And add here the check: https://github.com/gioblu/PJON/blob/master/src/PJON.h#L478
```cpp
if((_mode != PJON_SIMPLEX) && (!_router || (_router && router_bus_id)))
```

xlfe, 2018-10-13:
> The bus id filtering does not occur in the case `_router` is set true. So if a packet with a different recipient's bus id is picked up by the device in router mode your suggested device id check afterwards would let a sync ack to be sent when it is not supposed to do so.

Ah yes - I see what you mean.

> I am not sure this is the natural behavior is expected from a switch/router, I would discuss further this potential change

I see. My assumption about natural behaviour would be - bus->set_router(true) only changes behaviour in terms of whether there is a check as to whether the packet is destined for this device. That implies that where the packet is intended for this device, nothing should change? At the moment when set_router(true) is called, the bus no longer sends a synchronous ack if it is request.

My assumption might be different from others - so of course it is up to you I think :)

fredilarsen, 2018-12-28:
@xlfe, @gioblu A little late to comment on this, but it is very simple to send a sync ACK from user code in a router mode device, so I feel that it is perhaps cleaner to say that router mode always make the device into a passive listener, leaving routing logic including ACKing to user code.

closed 2021-03-22 by gioblu