gioblu/PJON #261: SAM C21 and SWBB

gnomathibus, 2018-10-29:
Hello everyone,
I'm trying to add a homemade board based on a SAM C21 (samc21e18a), the program compiles without error, I then try the program networkanalysis with an arduino uno as a transmitter and my sam c21 as a receiver and I receive the message busy for 10sec and after always fail ... I tested with the default timing and also the timing of arduino zero which is the most similar to the c21 but even pb.

gioblu, 2018-10-29:
CIao @gnomathibus try to add a 1 mega ohm pull down resistor, the busy result should disappear.
About the timing, if you have an oscilloscope you can measure the duration of bits of both devices and look for discrepancies. If there are any you can modify `SWBB_BIT_WIDTH`  and `SWBB_BIT_SPACER`  on one side to obtain equal results. Once done run tests tweaking the `SWBB_READ_DELAY` value on receiver side until communication quality is optimal.

gnomathibus, 2018-10-30:
hi gnoss, thank you for your answer, I just tried with a resistance of 1M and I have not busy, and the reception of data is still fail, I will try to find an oscilloscope but it is not won for me

gnomathibus, 2018-11-02:
Hi, good news, I managed to find an oscilloscope, I'm not an expert with the use of it but I learn quickly, the first steps I made is with the sketch BlinkTest transmitter independently on each card to compare the results. I have no idea why I do not find the same result?

first screenshot: arduino uno and the second SAM C21 :
[image]
[image]

gnomathibus, 2018-11-04:
Salut @gioblu thx for your help, I added Serial.println ((char *) string); and the result is the same whether it is the arduino UNO or the SAM C21: ", ?? 0-B⸮", I also built a second board SAM C21 and the sketch Blink does not work either between the two boards . I want to say that I have more arduino based atmega 328 and I have no problem with the sketch blink, if there is need I can do other measurement with the ocislloscope ...

gioblu, 2018-11-04:
Ciao @gnomathibus it's a pleasure, if you can take some other pic of the 2 signals could help, it is quite strange. Be sure to use the same program in both cases.

gnomathibus, 2018-11-05:
Slt  @gioblu , I attached a CSV image for more precision. 

[dump scope PJON BLINK TRANSMITTER CSV.zip](https://github.com/gioblu/PJON/files/2548306/dump.scope.PJON.BLINK.TRANSMITTER.CSV.zip)

another note, for the port of SAM C21 with arduino I use the version of mattairtech [https://github.com/mattairtech/ArduinoCore-samd](url), maybe the problem comes from here, and to finish the clock source used with the SAM C21 is an external crystal 32.768KHZ

`the code used on each board:`
#include <PJON.h>

// <Strategy name> bus(selected device id)
PJON<SoftwareBitBang> bus(45);

void setup() {
  bus.strategy.set_pin(12);
  bus.begin();
  bus.send_repeatedly(44, "B", 1, 1000000); // Send B to device 44 every second
}

void loop() {
  bus.update();
};`
**result with Arduino Duemilanove** 
[image]
**result with SAM C21**
[image]

gioblu, 2018-11-05:
Ciao @gnomathibus the problem maybe is here:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/SoftwareBitBang.h#L206
Could you print the result of `b & mask` on both devices?

gioblu, 2018-11-05:
@gnomathibus does the result change editing:
```cpp
PJON_IO_WRITE(_output_pin, b & mask);
// In
PJON_IO_WRITE(_output_pin, (b & mask) ? HIGH : LOW);
```
?

gnomathibus, 2018-11-05:
same result after test

gioblu, 2019-03-23:
Ciao @gnomathibus, sorry for the very late follow up.
I suspect the problem may be the type `char` used for data, that I would say is not portable, although it behaves the same in most cases: https://github.com/gioblu/PJON/blob/master/src/PJON.h#L162

> `char`  Smallest addressable unit of the machine that can contain basic character set. It is an integer type. Actual type can be either signed or unsigned. 

> `signed char` Of the same size as char, but guaranteed to be signed. Capable of containing at least the [−127, +127] 

If the sam behaves differently here, it may corrupt data.
Thank you for the report.

gnomathibus, 2019-03-23:
Salut @gioblu, thank you again for the interest in my problem, I do not have much time now but as soon as I have I look, meanwhile I use arduino pro mini instead, my project is to replace my current home automation installation by rail modules with PJON inside :wink:  that I built. The interest of the SAM C21 for me is more ram and functions like the integrated agenda and he operates in 5v.

here are some pictures of my PJON 40 modules in total it should interest you:
static relay, mechanical relay, light dimmer and ethernet

[image]
[image]

gioblu, 2019-03-23:
Ciao @gnomathibus those modules are really beautiful! Do you use gitter? 
Would be nice to have a chat! Compliments again!

gnomathibus, 2019-03-25:
Hi @gioblu, thank you for your compliments about my modules I joined the group on gitter :smiley:
I made this test on the SAM C21: "char d = -54" and the result gives 202 so the SAM C21 does not support the signed char, but what I do not understand is that the SAM D21 (arduino zero) works in SWBB while the SAM C21 and SAM D21 are very simililaires and I found this post: 

- [https://forum.arduino.cc/index.php?topic=399332.0](https://forum.arduino.cc/index.php?topic=399332.0)
 and here 
- [https://github.com/arduino/reference-en/pull/527](https://github.com/arduino/reference-en/pull/527)

 which indicates that arduino zero has the same problem.

gioblu, 2019-04-02:
Ciao @gnomathibus I have wrote you on gitter.
I agree, the same problem occurs on the SAM D21 and I suspect the switch to `uint8_t` will solve the issue.

gioblu, 2019-04-11:
Ciao @gnomathibus I have updated the codebase to handle `uint8_t` buffers and `const void *` parameters for the send functions, that should avoid the compatibility issue we have identified. I have ordered a some SAM D21 but they will need a couple of weeks to arrive. Could you give a try to master?

Thank you for your support.

gnomathibus, 2019-04-14:
Sorry for my late return (I do not have much time right now), I tested with the new pjon modifications but unfortunately does not change anything. For the moment I did the test only between the sam C21 and an arduino uno.

gioblu, 2019-05-31:
Ciao @gnomathibus I have run some experiments on SAMD21 and all works fine although SoftwareBitBang needed a minor timing update. Data is transmitted correctly.

Do you have any suggestions for a development/breakout board that uses  SAMC21?