gioblu/PJON #267: Inconsistency between packet_overhead and compose_packet treatment where ACK_MODE=1 and TX_INFO=0

xlfe, 2018-11-30:
Consider the case where PJON_ACK_MODE_BIT is set but PJON_TX_INFO_BIT is not

https://github.com/gioblu/PJON/blob/1d26b2784ac224fd24f20057baa58c0892ec0afd/src/PJON.h#L362-L382

Then here 

https://github.com/gioblu/PJON/blob/1d26b2784ac224fd24f20057baa58c0892ec0afd/src/PJON.h#L185-L188

The first function adds two bytes to the calculated overhead to accomodate the packet_id (because even if PJON_PACKET_ID_BIT is not set, PJON_ACK_MODE_BIT is set), but then when the packet is actually being composed, the logic changes and both PJON_ACK_MODE_BIT and PJON_TX_INFO_BIT must be set, and so a packet ID is not added to the packet.

xlfe, 2018-12-02:
I think the fix would be to change the calculation of packet overhead?

```diff
  + ( 
           ( 
-             (header & PJON_ACK_MODE_BIT) || 
+             ((header & PJON_ACK_MODE_BIT) && (header & PJON_TX_INFO_BIT)) || 
             (header & PJON_PACKET_ID_BIT) 
           ) ? 2 : 0 
         ) 
```

xlfe, 2018-12-02:
For information

I discovered this because I am writing a re-implementation of [PJON ThroughSerial in clojure](https://github.com/xlfe/PJON-clojure) - and I wanted to have [tests for every type of packet](https://github.com/xlfe/PJON-clojure/blob/master/test/jonure/autogen_test.clj) so I wrote a test generator using [added that to PJON-cython](https://github.com/xlfe/PJON-cython/blob/master/other/make-packets.py) which uncovered the weird case where tx-info is false and async-ack is true and packed-id is false...

gioblu, 2018-12-02:
Ciao @xlfe thank you very much for reporting this, and sorry for the late answer.
For sure the change you suggest would fix the inconsistency in the overhead calculation.
I observed and starred already PJON-clojure a couple of days ago compliments for this great work 🥇 
clojure at a first glance seems pure madness O_O never worked with that, but as soon as a sintetic README with some basic instructions will be available I will give it a try and report to you back after testing.

gioblu, 2018-12-02:
Don't mind to make a pull request @xlfe I will make the change and relate it to this issue.

xlfe, 2018-12-02:
> Don't mind to make a pull request @xlfe I will make the change and relate it to this issue.
> Thank you again.

Thanks @gioblu 

Yes clojure is an interesting language - at first it seems like pure madness - but then the more you get to know it the less it seems like madness and the more every other language does! :D LOL

Once I have PJON-clojure working properly (including a readme, etc) I will open an issue for it to be added to you list of PJON related projects - but yes it's not ready for real world use (yet)

gioblu, 2018-12-03:
Ciao @xlfe fixed, thank you again for raising this.

closed 2018-12-03 by gioblu