gioblu/PJON #274: [TSA] performance regression from 957afa46a2d0889b5220b4a2397f425a560039de and #271

xlfe, 2019-01-06:
I noticed a regression in performance between ESP32 using ThroughSerialAsync and PJON-cython after updating to the latest PJON version.

Using git bisect I was able to narrow it down to 957afa46a2d0889b5220b4a2397f425a560039de

If I run a [simple benchmark](https://github.com/xlfe/pjon-serial-benchmarks) which sends packets back and forth performance goes from ~32pps down to <6 pps after that commit.

I need to do a bit more work to figure out what in the commit actually impacted the performance though

gioblu, 2019-01-06:
ciao @xlfe thank you very much for reporting.
I suspect the speed change you see is provoked by the change I have done and you link above.
Before that change TSA was not waiting to detect the channel free for a duration longer than the response timeout (this procedure is specified in TSDL and is designed to avoid any possible collisions of a third device trying to start transmitting in between a packet and an ack). I see in your code that you have set a response timeout of 50 millis, so consider that before transmitting each device will wait for 50 millis and will start only if nothing was received in that time. So, if each exchange requires 50 millis of wait before transmission and around 50 millis are required for the ack to arrive, the theoretical maximum packet rate is 10 - the effective time used to transmit and execute the code. 

Obviously the collision avoidance described in TSDL is required only in multi-master, if using the master-slave, request-response procedure `TSA_TIME_IN` https://github.com/gioblu/PJON/blob/master/src/strategies/ThroughSerialAsync/Timing.h#L48 can be predefined to a value of `0`.

gioblu, 2019-01-06:
Ah @xlfe probably a quick test could be to predefine `TSA_TIME_IN` to 0 and see if you get ~32pps.

xlfe, 2019-01-07:
@gioblu yes adding #define TSA_TIME_IN 0 fixed the regression. Thanks for your quick response!

closed 2019-01-07 by xlfe

gioblu, 2019-01-08:
Ciao @xlfe thank you for your support.
I will release soon v11.2 so testing is more than welcome 👍