gioblu/PJON #277: ESP-NOW strategy support for esp8266

dominikborkowski, 2019-02-09:
Hi,
I was wondering if PJON's support for ESP-NOW could be extended to the esp8266 platform. On the surface it appears it may be currently available for the esp32.

Quick test using platformIO and sample code from  https://github.com/gioblu/PJON/blob/master/examples/ESP32/ESPNOW/PingPongArduinoIDE/Device1/Device1.ino I get errors such as:

```
.piolibdeps/PJON_ID245/src/strategies/ESPNOW/../../interfaces/ARDUINO/ESPNOWHelper.h:4:4: error: #error "ESP32 constant is not defined."
```

I can define said constant, but I'm not sure what the repercussions would be. Then we have another error regarding support for freertos library. Not sure which FreeRTOS library it expects:

```
.piolibdeps/PJON_ID245/src/strategies/ESPNOW/../../interfaces/ARDUINO/ESPNOWHelper.h:12:31: fatal error: freertos/FreeRTOS.h: No such file or directory
```

Using the sample code above on esp8266 one has to change

```
#include <WiFi.h> to
```

```
#include <ESP8266WiFi.h>
```

Thanks!

gioblu, 2019-02-09:
Ciao @dominikborkowski thank you for your support.

The first error may be fixed changing the condition as follows:
```cpp
#if !defined(ESP32) || !defined(ESP8266)
```
If you are using ESP8266 the constant should be defined and so the error should be avoided.

I am not sure about the second question, @xlfe (the creator of ESPNOW strategy), what do you think?

Changing the `#include` on esp8266 are you able to run the example successfully?
Thank you again for reporting this.

dominikborkowski, 2019-02-10:
Thanks for your prompt reply. I'm not sure why, but that logical OR doesn't seem to work. I can use simple
```
#if !defined(ESP8266)
```
or 
```
#ifndef ESP8266
```
and the code compiles just fine. As soon as I add the logical OR, regardless of order, it doesn't seem to work:

```
#if !defined(ESP8266) || !defined(ESP32)
    #error "ESP8266 or ESP32 constant is not defined."
#endif
```

```
from src/main.cpp:20:
.piolibdeps/PJON_ID245/src/strategies/ESPNOW/../../interfaces/ARDUINO/ESPNOWHelper.h:8:6: error: #error "ESP8266 or ESP32 constant is not defined."
#error "ESP8266 or ESP32 constant is not defined."
^
```

With regards to FreeRTOS, I'm wondering what's the applicable library? Presumably this should have been set up with platformIO as a dependency. PlatformIO lists quite a few FreeRTOS libs, none of them seem to be applicable to ESP8266: https://platformio.org/lib/search?query=freertos

Is it possible this is requiring 'esp8266-rtos-sdk' framework in platformIO as opposed to 'arduino'? 

Full details:

```
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <PJON> 11.2
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
Compiling .pioenvs/d1_mini/src/main.cpp.o
Generating LD script .pioenvs/d1_mini/ld/eagle.app.v6.common.ld
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFiAP.cpp.o
In file included from .piolibdeps/PJON_ID245/src/strategies/ESPNOW/ESPNOW.h:23:0,
from .piolibdeps/PJON_ID245/src/strategies/PJON_Strategies.h:57,
from .piolibdeps/PJON_ID245/src/PJON.h:96,
from src/main.cpp:21:
.piolibdeps/PJON_ID245/src/strategies/ESPNOW/../../interfaces/ARDUINO/ESPNOWHelper.h:16:31: fatal error: freertos/FreeRTOS.h: No such file or directory
#include "freertos/FreeRTOS.h"
^
compilation terminated.
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/d1_mini/lib09e/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
*** [.pioenvs/d1_mini/src/main.cpp.o] Error 1
================================================================ [ERROR] Took 1.66 seconds ================================================================
The terminal process terminated with exit code: 1
```

Platform:
```
espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Home: http://platformio.org/platforms/espressif8266
Frameworks: arduino, esp8266-nonos-sdk, esp8266-rtos-sdk, simba
Packages: framework-esp8266-rtos-sdk, toolchain-xtensa, framework-simba, tool-esptool, tool-mkspiffs, tool-espotapy, framework-arduinoespressif8266, sdk-esp8266, framework-esp8266-nonos-sdk
Version: 1.8.0
```

dominikborkowski, 2019-02-10:
Doh! This should have been the right incantation, logical AND not OR:

```
#if !defined(ESP8266) && !defined(ESP32)
    #error "ESP8266 or ESP32 constant is not defined."
#endif
```

dominikborkowski, 2019-02-10:
After further digging, it appears that ESP-NOW strategy leverages esp-idf, which is esp32 specific. I guess that answers the original question, whether this could be used on esp8266. It would be interesting to see if this strategy could be implemented eventually to support this simpler platform.

Thanks!

gioblu, 2019-02-10:
Ciao @dominikborkowski I am sorry, my stupidity to answer when there was not enough available time to develop a correct reply. I hope you did not lost of time because of that.

Indeed the logical OR is not the right way to do it, although also the AND is not what we are looking for considering that the error should be thrown if also just one of the two constants is defined, a way to solve that is to define 2 conditions, also to keep what is happening clear.
```cpp
#if !defined(ESP32)
    #error "ESP32 constant is not defined."
#endif

#if !defined(ESP8266)
    #error "ESP8266 constant is not defined."
#endif
```
I agree with you @dominikborkowski that woud be nice to have ESP8266 compatibility.
I will study further and see what can be done, obviously if you have any suggestion feel free to share :).

Thank you for your support.

xlfe, 2019-02-12:
I think it is possible - see https://github.com/yoursunny/WifiEspNow/tree/master/src for an example - I've got a few things on my plate so if anyone wants to give it a god they should go ahead and not wait for me to try!

dominikborkowski, 2019-02-13:
Thank you kindly for your responses. Unfortunately, writing that is currently beyond the scope of my experience with C/C++ and familiarity with PJON itself. However, I'd be more than happy to pitch in by testing/debugging if such port ever appears. Cheers!

gioblu, 2019-02-16:
Ciao @xlfe thank you for pointing out WifiEspNow, it deserves for sure some time to be studied, and compliments for the new PJON-cython release.

ulno, 2019-09-08:
Just wondering if any progress was achieved on the PJON ESP8266 ESPNOW implementation? Has anybody made some more experiments to have an idea how complex it would be?

gioblu, 2019-09-09:
Ciao @ulno, @dominikborkowski after a lot of work and thanks to the support of the community v12.0 will be soon released. After that, I will focus, solve this issue and add the compatibility. I am now sadly stuck with the release related hassle and I cannot focus fully on this although looking at [this library](https://github.com/yoursunny/WifiEspNow/blob/master/src/WifiEspNow.cpp) that has support for both ESP8266 and ESP32, it seems quite straight-forward. The ESPNOW strategy would need minimal changes to support both. I can start the work on this in the first week of october, if some of you has the will to work on this, all support is wellcome :)

happytm, 2019-12-23:
Is ESPNow protocol implemented yet on ESP8266 ?

ulno, 2020-02-06:
In my IoT class last year, we have used and run some tests with ESPNow, both on esp32 and esp8266 and got in spite of common myths surrounding ESPNow on the ESP32 the same result (only about 100m line of sight connections - though both platforms worked well).
We used the same reference mentioned before as base: https://github.com/yoursunny/WifiEspNow

Is anybody else taking a look at this? If not, I will see if I can make it work on PJON.

gioblu, 2020-02-07:
Ciao @ulno that would be a great help :) 
Thank you for your support.

Strix-CZ, 2020-08-09:
I have just created a pull request to introduce the support. I will be glad if you review the change and help me improve it. Also, I am not sure if I broke ESP32 support, because I only have ESP8266s laying around...

https://github.com/gioblu/PJON/pull/361