gioblu/PJON #278: PJON.h compile error on ESP8266 (bitbang)

canDry, 2019-02-09:
When I try to compile the ESP8266 software bitbang example (using any ESP8266 board - but I'm specifically trying it with a Wemos D1 mini) it fails with:

```
.../Arduino/libraries/PJON/src/strategies/LocalUDP/../../interfaces/ARDUINO/UDPHelper_ARDUINO.h:87:19: error: ambiguous overload for 'operator~' (operand type is 'IPAddress')
     broadcastIp = ~WiFi.subnetMask() | WiFi.localIP();
```
Using library PJON at version 11.2
Using library ESP8266WiFi at version 1.0

If I switch the board to ESP32 it compiles fine - so I may end up just using an ESP32, but am wondering what I'm doing wrong.

gioblu, 2019-02-09:
Ciao @canDry try to include only SoftwareBitBang strategy using `#define PJON_INCLUDE_SWBB true`

canDry, 2019-02-09:
> Ciao @canDry try to include only SoftwareBitBang strategy using `#define PJON_INCLUDE_SWBB true`

Added... recompiled.  Same result.

FYI, simply trying to compile this example:
https://github.com/gioblu/PJON/blob/master/examples/ESP8266/Local/SoftwareBitBang/SendAndReceive/ESP8266/ESP8266.ino

gioblu, 2019-02-09:
ciao @canDry thank you for reporting this and for linking the file, this seems strange, as described here:
https://github.com/gioblu/PJON/blob/master/documentation/configuration.md#data-link-configuration
if you define `PJON_INCLUDE_SWBB` **before** PJON.h inclusion the UDPHelper error should disappear.

gioblu, 2019-02-09:
I will make some tests and try to resolve the error. Thank you @canDry

canDry, 2019-02-09:
> if you define `PJON_INCLUDE_SWBB` **before** PJON.h inclusion the UDPHelper error should disappear.

Tip: never code before finishing your first coffee in the morning.  I added the define after the include.  Adding it before the include does indeed skip the issue causing the error!
Thanks so much for the quick response.

closed 2019-03-04 by canDry

mercurio04, 2019-05-02:
hi, I'm having this error.
I guess this could be because IPv6 is now internally supported by ESP8266 and it can't make sure that it's an IPv4 address for this operation.
my correction was: 
broadcastIp = ~uint32_t(WiFi.subnetMask()) | uint32_t(WiFi.gatewayIP());

gioblu, 2019-05-02:
Ciao @mercurio04 thank you very much for your report.
@fredilarsen have recently contributed with the same [fix](https://github.com/gioblu/PJON/commit/d0d34d5d42071dc022a7ffe8ba0ebaa60ae15a63) now included in master.

Happy tinkering!