gioblu/PJON #280: PJONMaster and Slave on TS

akadlec, 2019-02-18:
Is it possible to use MasterSlave with other strategies? I am using TS strategy over RS485. Right now esp8266 as master and uno as slave. When i try to use Master Slave, i get compile error, so i edited both library files:
```
template<typename Strategy = SoftwareBitBang>
```
to
```
template<typename Strategy>
```
(for slave PJONSlave.h)

but my master on ESP8266 is crashing

gioblu, 2019-02-21:
Ciao @akadlec `PJONMaster` and `PJONSlave` classes have been developed and tested with SoftwareBitBang strategy and Arduino Uno, Nano, Mega, Leonardo ecc.

I am now working on those classes to enable new features and simplify their codebase.
If you are able to gather any info about the crash that would be helpful.

I will in any case work on the ESP compatibility before the release.
Thank you for pointing out this and sorry for my late response.

gioblu, 2019-02-21:
Thanks also for exposing the outdated template declaration there.

gioblu, 2019-02-21:
Ciao @akadlec looking back at the example, maybe the definition of [`PJON_RECEIVE_WHILE_SENDING_BLOCKING`](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Network/SoftwareBitBang/MasterSlaveAutoAddressing/Master/Master.ino#L4)
freezes the ESP, could you give it a try avoiding its definition in master's and slave's program?
Thank you very much for your support.

akadlec, 2019-02-21:
Oh, thanks for your reply. I took your main parts from master/slave and reimplemented it into project i am working on. I thing one issue is in master when it boot up, there is loop which is looking for slaves `list_ids();` when i disable it, esp will boot normally

gioblu, 2019-04-01:
Ciao @akadlec I have updated the dynamic addressing implementation and specification. 
Although for now it does not handle device addresses collisions it is much more reliable and efficient.

gioblu, 2019-04-07:
I will close here @akadlec, feel free to re-open if required.
Happy tinkering!

closed 2019-04-07 by gioblu