gioblu/PJON #281: Problem with send_repeatedly - not able to reset

akadlec, 2019-02-23:
I am reimplementing MasterSlave to TS and RS485. Everything is working well. I use `send_repeatedly` as you implemented in your masterslave. But when init fails and slave negate for new address, master keeps sending repeatedly packet, even if i call clearing `remove_all_packets` for all slaves.

gioblu, 2019-02-23:
Ciao @akadlec that may mean that you are for some reason loosing the pointer to a packet in your code for example overwriting another or deleting its value without removing the packet.

I am working on a new version of the dynamic addressing specification and implementation, which does not need the packets buffer and does not use anymore that broadcast scheme, that on data-link like RS485/serial may not be the best solution. I will soon start to release the changes.

akadlec, 2019-02-25:
ok, it could be. i am closing it

closed 2019-02-25 by akadlec