gioblu/PJON #283: Feature: Dynamic addressing in Shared mode

mxbranson, 2019-03-18:
Hi,
Awesome to see this developing along.
I have a situation where I can have 255+ radios talking to one gateway which require all require dynamic addressing.
Is this possible using shared mode with the gateway/master assigning a free bus/ID pair to the slave on request.
The bus id should be selected from an array variable of available gateways to avoid (global) collisions.

Thanks again!

gioblu, 2019-03-18:
Ciao @mxbranson thank you.
The `PJONMaster` and `PJONSlave` classes can operate in both local and shared mode already.
See this example (that uses SoftwareBitBang but can be easily adapted to your needs): https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Network/SoftwareBitBang/MasterSlaveAutoAddressing

The classes are designed to let up to 253 slave to be connected to a master in shared mode that owns a bus id. All communication between slaves and master occur in shared mode on a certain bus id. Although auto-discovery is not implemented and the addressing scheme is based on randomly generated addresses that are not that useful most use cases, but essential in particular edge cases.

I am now working on a new version of the dynamic addressing that supports auto-discovery and a more useful addressing scheme that will enable users to reach a given physical device using a unique address not caring at all about device and bus id. A sort of MAC, although is on top of the network addresses instead of below (as specified in the osi).

If you are interested to discuss about it and see if the new scheme fits your requirements feel free to join our gitter chat.

mxbranson, 2019-03-19:
That sounds great! I'm thinking how PJON could be expanded into a open source cell network with codec 2's 700 bit audio codec.
Joining gitter.

closed 2019-03-19 by mxbranson