gioblu/PJON #284: Missing header file in 11.2

Mirodin, 2019-03-18:
Forgive me if this is a problem with PlatformIO but I thought I might find help here since it is an issue with PJON.

I wanted to get my feet wet with microcontrollers and stuff and found PJON quite intriguing for networking. However I have PlatformIO (v.3.6.0) running on my system and installed PJON via their CLI which pulls PJON 11.2.

However when I want to upload my sketch to my Arduino board, the build process yells at me for missing header files - at least I think that is what is missing, I am still in early stages of learning C universe. So far I have only experience in scripting languages therefore debugging compilation is still quite alien to me. Again sorry if I am wrong here and this is a PlatformIO issue.

```
> platformio run
Processing uno (platform: atmelavr; board: uno; framework: arduino)
------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Arduino Uno
SYSTEM: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Converting main.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PJON> 11.2
Compiling .pioenvs/uno/src/main.ino.cpp.o
In file included from .piolibdeps/PJON_ID245/src/strategies/EthernetTCP/EthernetLink.h:129:0,
from .piolibdeps/PJON_ID245/src/strategies/EthernetTCP/EthernetTCP.h:24,
from .piolibdeps/PJON_ID245/src/strategies/PJON_Strategies.h:76,
from .piolibdeps/PJON_ID245/src/PJON.h:96,
from /home/mirodin/projects/buddy/uno/src/main.ino:1:
.piolibdeps/PJON_ID245/src/strategies/EthernetTCP/../../interfaces/ARDUINO/TCPHelper_ARDUINO.h:16:26: fatal error: Ethernet.h: No such file or directory

******************************************************************
* Looking for Ethernet.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Ethernet.h"
* Web  > https://platformio.org/lib/search?query=header:Ethernet.h
*
******************************************************************

compilation terminated.
*** [.pioenvs/uno/src/main.ino.cpp.o] Error 1
================= [ERROR] Took 0.48 seconds =================
```

gioblu, 2019-03-18:
Ciao @Mirodin thank you very much for your report.
I am sadly not an expert of PlatformIO's inner workings, although it seems it is not able to find the Ethernet library. It suggests to look for this missing dependency here:
https://platformio.org/lib/search?query=header:Ethernet.h

Which strategy is used in the program you are trying to compile?

Mirodin, 2019-03-18:
Hi Giovanni,

thank you for answering me. I am quite new to PlatformIO myself as well but it seems to be the only somewhat free solution I have other than the Arduino IDE. Whilst debugging I removed line by line and in the end it came down to this which causes the issue:

```
#include <PJON.h>

PJON<SoftwareBitBang> bus(1);
```

I tried the Ethernet library that was most popular as I figured it might be the most likely candidate but that caused another error message:

```
> platformio run
Processing uno (platform: atmelavr; board: uno; framework: arduino)
------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Arduino Uno
SYSTEM: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Converting main.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PJON> 11.2
|   |-- <Ethernet> 2.0.0
|   |   |-- <SPI> 1.0
Compiling .pioenvs/uno/src/main.ino.cpp.o
Archiving .pioenvs/uno/libf6d/libSPI.a
Indexing .pioenvs/uno/libf6d/libSPI.a
Compiling .pioenvs/uno/libf76/Ethernet_ID872/Dhcp.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/Dns.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/Ethernet.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/EthernetClient.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/EthernetServer.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/EthernetUdp.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/socket.cpp.o
Compiling .pioenvs/uno/libf76/Ethernet_ID872/utility/w5100.cpp.o
Archiving .pioenvs/uno/libf76/libEthernet_ID872.a
Indexing .pioenvs/uno/libf76/libEthernet_ID872.a
Linking .pioenvs/uno/firmware.elf
/tmp/cc5vw9Bz.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x174): undefined reference to `setup'
<artificial>:(.text.startup+0x17c): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pioenvs/uno/firmware.elf] Error 1
================= [ERROR] Took 0.91 seconds =================
```

gioblu, 2019-03-18:
Ciao @Mirodin thank you for your quick response.
It seems he does not like something in the program's code?

```
/tmp/cc5vw9Bz.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x174): undefined reference to `setup'
<artificial>:(.text.startup+0x17c): undefined reference to `loop'
```

Could you share it here?

Mirodin, 2019-03-18:
I did above already, it is just those two lines I had after removing line by line and checking if the error persists:
```
#include <PJON.h>

PJON<SoftwareBitBang> bus(1);
```

gioblu, 2019-03-18:
Ciao @Mirodin thank you for your answer, I have understand your point.
A quick solution is to define `PJON_INCLUDE_SWBB` before `#include <PJON.h>
That should avoid the dependency problem.

For sure I must add the `dependencies` in `library.json` thank you for pointing out.
Will try to study PlatformIO more in detail as soon as I will have some free time.

Mirodin, 2019-03-19:
Hi Giovanni,

thank you for your help, that quick fix made my compilations pass. Thank you for this awesome peace of software as well, I will see if I can help when I get more used to the whole arduino world.

dtony, 2020-02-29:
Hi, I had the same problem with PlatformIO and I was able to solve it by removing the global installation of the Ethernet library in PlatformIO. I removed the Ethernet library directory in ~/.platformio/lib/

gioblu, 2020-06-14:
Finally solved by https://github.com/gioblu/PJON/commit/d2585c3f0fbebb5b2cde26301f3520b50816e255 thank you for reporting.

closed 2020-06-14 by gioblu