gioblu/PJON #288: ThroughSerialAsync full buffer handling

chestwood96, 2019-04-13:
It appears that Serial.write returns but does not actually send anything if the buffer is full so big packets get corrupted.

I fixed this temporarily using a while loop but it probably needs a better solution (avoiding infinite loop and all that).

```
void send_byte(uint8_t b) {
	while (PJON_SERIAL_WRITE(serial, b) != 1) {}
};
```

gioblu, 2019-04-13:
Ciao @chestwood96 thank you very much for your report. Absolutely, handling the full buffer case is necessary, I will work it through properly as soon as possible.

Thank you again :)

gioblu, 2019-04-16:
Ciao @chestwood96 I have added the fix, thank you very much.

I use the constants `TS_BYTE_TIME_OUT` and `TSA_BYTE_TIME_OUT` as maximum time dedicated to attempt to send a byte (they were also used already as the maximum time dedicated to a byte reception ).

chestwood96, 2019-04-16:
Thanks for the fix. I am currently writing a prototype implementation of the protocol in dotent core to use in my higher level applications and this made me question my sanity before I figured it out.  I'll let you know if I find anything else.

gioblu, 2019-04-16:
Cool, I am sure many other users will appreciate that.
Thank you again for your support.

closed 2019-04-16 by gioblu