gioblu/PJON #289: Debugging PJON connection between ESP8266 (nodemcu) and Attiny85

henrik-muehe, 2019-04-17:
I am not sure if this is a great forum to ask this question, let me know if I should move elsewhere.

I am trying to communicate between an Attiny85 and a nodemcu. SoftwareBitBang works between Attiny85 and an Arduino Uno. It also works between two Nodemcus but not between the ATtiny85 and the Nodemcu. I am running all chips (except the Uno) at 3.3 for this experiment. 

I look at the signal on the wire with an oscilloscope and the signals for writing a single byte look pretty much exactly the same between the attiny85 and the ESP except for the part where the sender sends the 1/4s high signal waiting for a response (and said response is missing / times out).

Any direction for debugging further would be appreciated. Is there code instrumentation that I could / should do? Are there known interop issues like this?

gioblu, 2019-04-17:
Ciao @henrik-muehe it seems it may be an issue related to the discrepancy in the duration of the 1/4 bit high emitted by transmitter and its expected duration in the receiver. 

To verify it, just run the same example disabling the synchronous acknowledgement on transmitter side.
If works fine, the issue must be caused by the discrepancy described above.

Let me know your the test results if you can, If this is the case I will fix it as soon as possible.

I have not identified other interop issues like this one, we are tackling a problem with SAM but it is not related to this issue.

Thank you for your report.

gioblu, 2019-04-17:
Ciao @henrik-muehe if the problem is unsuccessful reception, on which side occurs? If you switch the sides still occurs?

gioblu, 2019-04-29:
As discussed on gitter, we have identified that the problem was in the use of the internal PPL clock, that is not supported for now. I think could be possible to define a slower mode that could work with the internal oscillator and its lower precision (if compared to an average external crystal oscillator). I will try to work out the slower mode in the next iteration. Thank you for reporting and happy tinkering.
Feel free to re-open the issue if required.

closed 2019-04-29 by gioblu