gioblu/PJON #292: InteractiveRouter not sending synchronous ack for packets directed at it

chestwood96, 2019-04-28:
It appears the InteractiveRouter does not send a synchronous ack when it receives the packet itself.

I have worked around that by sending it myself right now but should it not do it itself?

gioblu, 2019-04-29:
Ciao @chestwood96 thank you very much.
I have discussed this with @fredilarsen on gitter, and we agree, the `InteractiveRouter` being a class that wants to implement the features of the router along with the ones of the device, it should send the synchronous acknowledge automatically as you suggest.

I will apply the change as soon as possible.
Thank you again.

chestwood96, 2019-04-30:
While you are at it it would be nice to have a regular PJON-Like send function for the interactive router.
The current one kind of requires you to either build the whole packet or know which bus to send to.

Should i make a separate issue for this?

gioblu, 2019-04-30:
Don't be preoccupied @chestwood96 I will add the overload to the functions you need, thank you for pointing out.

gioblu, 2019-05-03:
Ciao @chestwood96 I have added the ack fix you have proposed.

@fredilarsen how would you implement the PJON-like send functions?

I think @chestwood96 is right to expect to find them, being the InteractiveRouter also a device, should provide the functions to send normally as a device, although those functions should leverage of the router features.

I have made some experiments, I suspect they should:
- If the bus and device id is found in the list of buses, send directly
- If not, if there is a default gateway, send to default gateway
- if not, should he do nothing or flood?

fredilarsen, 2019-05-10:
@gioblu, @chestwood96 There is a PJONINteractiveRouter::send_packet which should do this:

```void send_packet(
    const uint8_t *payload,
    uint16_t length,
    const PJON_Packet_Info &packet_info
  )```

It requires you to declare a local Packet_Info object, but that only contains the parameters needed for sending anyhow (from_id, to_id, potential bus ids etc).

chestwood96, 2019-05-11:
I was actually using `router.get_bus(router.get_callback_bus()) ` for the testing.

I kind of looked at PJON_Packet_Info like something pjon internal I should not be messing with from the outside.

gioblu, 2019-05-14:
@chestwood96 yes the `PJON_Packet_info` is used by PJON internally and to let the user access to packet's meta-data within the receiver function callback. I agree with you that could be more straight-forward to expose to users the same `send` function parameters the `PJON` class exposes. That may avoid mistakes and confusion. What do you think about this @fredilarsen ?

gioblu, 2019-05-14:
To be clear for the `send` functions a list of parameters is used instead of passing the `PJON_Packet_Info` struct to minimize the amount of memory overhead on setups that use the least amount of features, for example ATtiny85 running in local, and to minimize the amount of code lines required to send a packet.

gioblu, 2019-08-11:
Ciao @chestwood96 at the end we understood that was necessary, as you correctly pointed out. Here is the commit that fixes it https://github.com/gioblu/PJON/commit/9fc81b3af01812d5712e14ac866b0a831710f7d1

Soon v12.0 will be released including this optimization. 
Thank you for raising this.

closed 2019-08-11 by gioblu