gioblu/PJON #293: Master.ino example is crashing on ESP82xx

jeffaries, 2019-05-02:
Hi, 
This is my first contribution and definitely not my last. I have discovered PJON a couple of weeks ago and I have decided to do some POC in the context of a project for bird breeders. I have successfully tested all aspects important to me and my project... It is a super great library.
However, I am facing an issue when I try to run a Master on an Esp8266 or Esp8285 (These are the only microcontrollers I have so far).
I have tuned a bit the Master.ino file (the right pin, #define PJON_INCLUDE_SWBB true, and my bus_id)
```
#define PJON_INCLUDE_SWBB true

/* This let the master receive a packet while trying to send another.
   Avoids network instability */
#define PJON_RECEIVE_WHILE_SENDING_BLOCKING true

#include <PJONMaster.h>

// Bus id definition
uint8_t bus_id[] = {0, 0, 1, 102};
// PJON object - The Master device id is fixed to PJON_MASTER_ID or 254
PJONMaster<SoftwareBitBang> bus(bus_id);

uint32_t t_millis;

void setup() {
  Serial.begin(115200);
  /* Let addressing procedure packets to be received by the receiver function
     to ease debugging or analysis */
  bus.debug = true;

  bus.strategy.set_pin(5);
  bus.set_receiver(receiver_function);
  bus.set_error(error_handler);
  bus.begin();
  /* Send a continuous greetings packet every secon
     to showcase the receiver function functionality if debug is active*/
  if(bus.debug)
    bus.send_repeatedly(PJON_BROADCAST, "Master says hi!", 15, 2500000);
  t_millis = millis();
};

void error_handler(uint8_t code, uint16_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("PJON error: connection lost with device id ");
    Serial.println((uint8_t)bus.packets[data].content[0], DEC);
  }
  if(code == PJON_ID_ACQUISITION_FAIL) {
    Serial.print("PJONMaster error: connection lost with slave id ");
    Serial.println(data, DEC);
  }
  if(code == PJON_DEVICES_BUFFER_FULL) {
    Serial.print("PJONMaster error: master devices' buffer is full with a length of ");
    Serial.println(data);
  }
};

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */

  // If debug is active prints addressing packet infromation
  if(packet_info.port == PJON_DYNAMIC_ADDRESSING_PORT) {
    uint32_t rid =
      (uint32_t)(payload[1]) << 24 |
      (uint32_t)(payload[2]) << 16 |
      (uint32_t)(payload[3]) <<  8 |
      (uint32_t)(payload[4]);
    Serial.print("Addressing request: ");
    Serial.print(payload[0]);
    Serial.print(" RID: ");
    Serial.print(rid);
  }

  // General packet data
  Serial.print(" Header: ");
  Serial.print(packet_info.header, BIN);
  Serial.print(" Length: ");
  Serial.print(length);
  Serial.print(" Sender id: ");
  Serial.print(packet_info.sender_id);

  // Packet content
  Serial.print(" Packet: ");
  for(uint8_t i = 0; i < length; i++) {
    Serial.print(payload[i]);
    Serial.print(" ");
  }
  Serial.print("Packets in buffer: ");
  Serial.println(bus.update());
};

void loop() {
  if(millis() - t_millis > 5000) {
    // Check if registered slaves are still present on the bus
    bus.check_slaves_presence();

    Serial.println("List of slaves known by master: ");
    for(uint8_t i = 0; i < PJON_MAX_DEVICES; i++) {
      if(bus.ids[i].state) {
        Serial.print(" - Device id: ");
        Serial.print(i + 1); // Shifted by one to avoid PJON_BROADCAST
        Serial.print(" Device rid: ");
        Serial.print(bus.ids[i].rid);
        Serial.println();
      }
    }
    Serial.println();
    Serial.flush();
    t_millis = millis();
  }
  bus.receive(5000);
  bus.update();
};
```

The rest is your code. 

When I launch the espxxxx... I get:
```
Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 01b0
3fffff20:  00000000 3ffee3f7 00000000 40202d41  
3fffff30:  00000000 3fffff70 3ffee3ee 3ffee7c4  
3fffff40:  3ffee3ec 0195363c 0003d090 402033dc  
3fffff50:  0172da5c 3ffee3f7 3ffee3ec 40203464  
3fffff60:  00000037 00000000 00000001 3fffefa0  
3fffff70:  402013cc 0000038f 0000038f 402013ef  
3fffff80:  000000fe 3ffee4ec 3ffee3ec 40203511  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee794 402043d0  
3fffffb0:  feefeffe feefeffe 3ffe84f8 40100a59  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```
That is a typical sever crash.

Do you have any idea what's going wrong?

Thank you very much!

Jeff

gioblu, 2019-05-02:
Ciao @jeffaries thank you very much for your compliments.
Yes, the old version of the dynamic addressing had issues with esp and if I remember correctly it was related to `check_slaves_presence`.

I have now developed dynamic addressing v3.0 that you can find in master, see its [specification](https://github.com/gioblu/PJON/blob/master/specification/PJON-dynamic-addressing-specification-v3.0.md), [documentation](https://github.com/gioblu/PJON/blob/master/documentation/addressing.md) and [implementation](https://github.com/gioblu/PJON/tree/master/src). The function that was crashing the esp has been removed so it should work, although I will run a test a soon as possible and report back results. 

Thank you for reporting this.

gioblu, 2019-05-02:
Sorry @jeffaries forgot to link the updated example: https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Network/SoftwareBitBang/MasterSlaveAutoAddressing

jeffaries, 2019-05-02:
Gracie Giovanni,

It is indeed working now ;)

I can continue my adventure with PJON.
FYI, I am testing with ESP8285... and even with ESP01 with the 1-wire on GPIO2 (GPIO0 is not working).

Note: Even the ESP8285, a DIYMORE model, accepts to work with other GPIOs.

Thank you again

Jeff

closed 2019-05-02 by jeffaries

gioblu, 2019-05-02:
Thank you for your detailed report @jeffaries 
Can I ask you on those new devices we have never tried, on which pin works and which speed you get from the NetworkAnalysis test? So I could add that to the compatibility matrix

jeffaries, 2019-05-13:
@gioblu , I will do it.
Do you need only ESP8285 in the network or one is enough? So far, I have only one, others are ordered and are on the road (the silk road). 
For ESP01, I have several, so I can launch a test.

gioblu, 2019-05-13:
Ciao @jeffaries thank you very much for your support to the project 👍 
ESP01 tests if you have at least 2 would be very useful (you could verify that ESP01 works between each other and with other supported devices)

To be sure all works fine I suggest you to run the `NetworkAnalysis` example both ways (EPS01 tx other rx and the contrary)