gioblu/PJON #294: EthernetTCP strategy, loosing responses

mercurio04, 2019-05-06:
Ciao!
I'm using platformio with esp8266 and windows 10.

I'm using the examples receiver and transmitter, (receiver in esp and transmitter in win) 

I'm loosing responses, like 50% of them. Using bus.reply
I'm using the example as it is in github

Is it the example outdated? 
 
Grazie Mille
Saludos de Argentina

gioblu, 2019-05-06:
Ciao @mercurio04 looking at the code I think the motivation why some replies fail may be because of the network and OS latency in between packet transmission and response reception. Try to change the `ETCP_IDLE_TIMEOUT` to a higher value:
```cpp
#define ETCP_IDLE_TIMEOUT 60000ul
```

fredilarsen, 2019-05-06:
It may be that using the ESP over WiFi introduce more latency than was present when the timeout value was selected, so I agree increasing the ETCP_IDLE_TIMEOUT is worth a try.

I will test EthernetTCP with the combination of ESP over WiFi and Windows when I get the time.

Do you need a one-to-one connection or do you want to create a many-to-many or master-slaves bus?
The DualUDP strategy that has been added after EthernetTCP may be easier (no IP addresses needed) if you want a many-to-many setup on a LAN, but still need to talk to devices on other network segments.

EthernetTCP can be run in the SINGLE_DIRECTION mode that makes it use two persistent sockets, more ideal for a bidirectional one-to-one connection with high speed and one-way firewall port forwarding if needed. See the "Surrogate" example if this is of interest, letting the WIndows process be "part of" a SWBB or other bus type by using a Surrogate (somewhat similar to a WiFi-to-SWBB Switch but with a dedicated higher speed link).

mercurio04, 2019-05-06:
I'm trying to create many to many connection. so I will try increasing the timeout. anyway, I will try also LocalUDP strategy, 
I will come to you with the results.
Gracias!

fredilarsen, 2019-05-10:
@mercurio04 I would rather recommend the DualUDP strategy instead of LocalUDP, as UDP broadcast does not work very well on the ESP cards for some reason. 
DualUDP uses UDP broadcasts initially to establish routes but will then go to directed sending which will perform better.

mercurio04, 2019-05-10:
I was trying DualUDP, and for an unkown reason i got responses for a few seconds, and then nothing else.   using wireshark, I can see the udp broadcast, still active but I'm not getting any responses. I tried also with an ESP32 as a receiver and ESP8266 as a transmitter, and the result is the same. but if I use the ESP32 in Access Point mode, the comunication works perfect. so.. just in case i will try with another wi fi router.
anyyway, thanks @fredilarsen for explain me the difference between LocalUDP and DualUDP.

fredilarsen, 2019-05-10:
@mercurio04 I my tests I used a pair of ESP8266, and later a pair os ESP32, connected as DHCP clients to my WiFi network (Netgear WiFI router as access point).

closed 2019-07-12 by gioblu