gioblu/PJON #295: SoftwareBitBang - One wire without common ground

jeffaries, 2019-05-13:
Hi everybody, I am looking for advices on the correct wiring to connect multiple devices (ATTiny85 based), organized in stars with own power supply (all with the same voltage 5V) to use with SoftwareBitBang (final topology: tree). Do I need to add an extra ground wire to the connection between stars or can I bridge their grounds? 
Isn't there a way to connect the simply connect these stars with one wire???

gioblu, 2019-05-13:
CIao @jeffaries if you have already a gound present at each node you can avoid to transport the ground along with the data line. Although, specially if the setup is vast and in close vicinity of AC wiring, it may be more cautious to transport the ground in parallel with the signal wire. 
See: https://github.com/gioblu/PJON/wiki/Protective-circuitry#avoid-ground-loop

jeffaries, 2019-05-13:
My latest tests show that without a common ground, the SBB is not working. So, how could it work without transporting the ground? 
The setup could potentially include 100 nodes but on a not so large distance (longest road would be max 20m). The 100 nodes would be grouped (max 8 nodes per group) in star networks powered with one power supply per star. The bus cable would bind these stars together (with a Master and up to 100 slaves). That's why I worry about a common ground.

When you mention "to transport the ground in parallel with the signal wire", where this ground line would come? From the Master?

gioblu, 2019-05-13:
Ciao @jeffaries I meant that If you have an existing ground cable in the vicinity of the nodes, you could avoid to transport the ground from the master to all its slaves. 

If you don't have an existing wiring to operate with, you can: 
- Connect a ground wire from one power supply to all its nodes 
- Connect the grounds of the power supplies together 

Consider that is generally suggested to feed a circuit with only one connection to ground.
Consider also to add protective circuitry: https://github.com/gioblu/PJON/wiki/Protective-circuitry
Here info for mitigating interference: https://github.com/gioblu/PJON/wiki/Mitigate-interference

jeffaries, 2019-05-13:
Thanks a lot for your help.

gioblu, 2019-05-13:
CIao @jeffaries thank you for your support to the project.
Feel free to use the gitter chat for direct support, I see you have wrote there but I was not online at the time.

closed 2019-05-21 by gioblu