gioblu/PJON #297: Tutorial Videos about PJON on YouTube

mauricebauer, 2019-05-28:
Hey there,

first: I am really thankful and impressed by your great work on PJON 👍 

Now the question: I am study Computer Science and I think the start with PJON could be easier. The documentation is great, but: It's a documentation. I think its purpose is to strictly define how the things work.
Is there a tutorial like guide on how to use if properly? I've seen you have some videos about PJON on YouTube, could you make some Video in a more Presentation-Style how to use PJON. I could imagine that would increase the popularity of PJON...

Thanks in advance and again: great work on PJON, keep up the work...
- Maurice

Sorry for my bad english, I am from Germany.

gioblu, 2019-05-28:
Ciao @mauricebauer thank you very much for your compliments, it is really a pleasure to know that this experimental work is becoming more and more used also in the real world :)

I agree with you about the necessity of a series of videos/tutorials presenting the protocols features and the more common use cases.

Adam of [AdamBuilds](https://www.youtube.com/channel/UClyUGbnXRe_YU-NLlAPHlDg) is working on is own test of the [SoftwareBitBang](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) strategy, so if you are interested keep an eye on his channel.

Thank you for your valuable suggestion, I will dedicate some time to work out decent video tutorials showcasing the protocol

gioblu, 2019-05-28:
When the Adam's video will be published I will add the links in the repo and related the commits to this issue. Thanks again.

mauricebauer, 2019-05-29:
I am really looking forward to this videos. I hope in the future the start with PJON would be as easy as possible for beginners and also for professionals (who want to start as fast as possible).
Maybe I will make some videos too in the future, at the moment I dont know.

So thanks again and I wish huge success with PJON in the future :)
PS: Is there a PC software to use as a "bus monitor" so you can see all messages of a (for example SoftwareBitBang Bus) Bus with all its header bits etc...? Something like this: 
https://www2.knx.org/media/img/ETS5/ETS-App-SDK-for-Monitor-Apps-in-ETS5.png

gioblu, 2019-06-08:
Ciao @mauricebauer here is the link to adam's work: https://www.youtube.com/watch?v=8okhWVkMnng

For sure something like that would be useful! It may be done using an Arduino that has both SoftwareBitBang and serial over usb, that shares with the computer the received data over the bus via serial.

closed 2019-06-13 by mauricebauer

gioblu, 2019-12-13:
Ciao @mauricebauer I have added also:
- AnalogSampling: https://www.youtube.com/watch?v=1BeGYMjg-DI
- SoftwareBitBang: https://www.youtube.com/watch?v=GWlhKD5lz5w