gioblu/PJON #298: ATTiny85 without external oscillator

jeffaries, 2019-05-29:
Hi, 
I am analyzing the requirements for my project. 
I would need a microcontroller that can have 3 Digital outputs + 1 Analog input + 1 GPIO for PJON SoftwareBitBang...
I am considering the ATTiny85, however, I read about the "suggestion" to use an external oscillator (https://github.com/gioblu/PJON/wiki/ATtiny-interfacing) ... this would jeopardise my project since I would have only 3 GPIOs available while I need 5.
what is the risk not to use an oscillator? What would be the consequence/limitations? Would there be any mitigation action that would allow me to use the ATTiny85 without oscillator with PJON?

Thank you

gioblu, 2019-05-29:
Ciao @jeffaries the problem is the inherent inaccuracy of the internal PPL clock.
I have never tried to calibrate although if I am not mistaken that can be done. 

What I have experimentally determined is that from the same batch attinies using the internal PPL clock can drift enough to break the PJDL binary timing and so not be able to communicate with each other. Using an external crystal oscillator solves the issue. 

Consider the ATtiny84 (that is supported already) if the problem is the amount of input ouput pins required, see http://ww1.microchip.com/downloads/en/devicedoc/doc8006.pdf

jeffaries, 2019-05-29:
Thank you very much for your answer. Since, it is not worth the risk, I will follow your advice and go for ATtiny84.

closed 2019-05-29 by jeffaries