gioblu/PJON #300: Add support for Particle Photon?

Seserie, 2019-06-10:
I use both Arduino and Particle Photons, and I would like to see PJON implemented into the Particle Library System, I am currently working my way through just copying it into the Web IDE but it would be much appreciated if support could be added! (I would love to help if there was anything I could do?) I've been using PJON for a bit but this is the first time I have actually posted in the PJON community!

gioblu, 2019-06-10:
Ciao @Seserie thank you for your support and welcome :). 
I will buy a couple of Particle Photon and see what I can do. 
I think this will come after release of v12.0 (soon).

How PJON behaved in your projects? 
Have you spotted any limitation/issue?

Thank you again for your feedback!

Seserie, 2019-06-11:
As of right now, I have gotten the Software BitBang to work! I love the Photon's as I am able to upload firmware to the device without having to connect it to my computer. 

I am trying to get the MasterAutoAddress to work on a Particle Photon with the Slave being some Arduino Nano's but I am getting the error of PJON_Packet_Info does not name a type... I have been trying to figure it out for a few days at this point...

gioblu, 2019-06-11:
Ciao @Seserie thank you very much for your feedback.
Which changes you have done to get it working on Particle Photon? 
Were you forced to define a dedicated timing to have compatibility with other supported boards?

Would you paste here the error you get for us to further study that?

Seserie, 2019-06-11:
So I had to copy and paste the source files: PJONMaster.h, PJON.h, PJONDefines.h, PJON_ARDUINO_INTERFACE.h (I left out the PJON_IO.h for Arduino as the ports are different and I am pretty sure that isn't required), PJON_CRC32.h, PJON_CRC8.h, PJON_Strategies.H, SoftwareBitBang.h, and Timing.h

The changes I made were:
Arduino Interface File, I commented out the Arduino Define, all the commands are the same between Arduino and Particle devices. So basically I am making PJON use the Arduino Interface as they use the same commands.

My changes for the Stratagies file were to comment out all the includes except for the SoftwareBitBang.h include. So none of the other stratagy files were used.

So as for the code for devices:

// This #include statement was automatically added by the Particle IDE.
#include <neopixel.h>

/* Manual synchronous acknowledge requires longer than default timeout
   Set higher duration if required */
#define SWBB_RESPONSE_TIMEOUT 5000
#include "Defines.h"
#include "PJONMaster.h"

// IMPORTANT: Set pixel COUNT, PIN and TYPE
#define PIXEL_PIN D7
#define PIXEL_COUNT 1
#define PIXEL_TYPE WS2812B

Adafruit_NeoPixel pixels(PIXEL_COUNT, PIXEL_PIN, PIXEL_TYPE);

// Bus id (unique bus identifier)
uint8_t bus_id[] = {0, 0, 0, 1};

// PJON object, device id is fixed to PJON_MASTER_ID or 254
PJONMaster<SoftwareBitBang> bus(bus_id);

uint32_t timmy;

void error_handler(uint8_t code, uint16_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("PJON error: connection lost with device id ");
    Serial.println((uint8_t)bus.packets[data].content[0], DEC);
    pixels.clear();
    pixels.setPixelColor(0, pixels.Color(255, 0, 0));
    pixels.show();   // Send the updated pixel colors to the hardware.
  }
  if(code == PJON_DEVICES_BUFFER_FULL) {
    Serial.print("PJONMaster error: master devices' buffer is full with a length of ");
    Serial.println(data);
    pixels.clear();
    pixels.setPixelColor(0, pixels.Color(255, 0, 0));
    pixels.show();   // Send the updated pixel colors to the hardware.
  }
};

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {

  // If debug is active prints addressing packet infromation
  if(packet_info.port == PJON_DYNAMIC_ADDRESSING_PORT) {
    Serial.print("Addressing request: ");
    Serial.print(payload[0]);
    Serial.print(" Device address: ");
    Serial.print(payload[1]);
    Serial.print(".");
    Serial.print(payload[2]);
    Serial.print(".");
    Serial.print(payload[3]);
    Serial.print(".");
    Serial.print(payload[4]);
    Serial.print(".");
    Serial.print(payload[5]);
  }

  // General packet data
  Serial.print(" Header: ");
  Serial.print(packet_info.header, BIN);
  Serial.print(" Length: ");
  Serial.print(length);
  Serial.print(" Sender id: ");
  Serial.print(packet_info.sender_id);

  // Packet content
  Serial.print(" Packet: ");
  for(uint8_t i = 0; i < length; i++) {
    Serial.print(payload[i]);
    Serial.print(" ");
  }
  Serial.print("Packets in buffer: ");
  Serial.println(bus.update());
};

void setup() {

  //Start Status LED
  pixels.begin();
  pixels.setPixelColor(0, pixels.Color(0, 0, 255));
  pixels.show();   // Send the updated pixel colors to the hardware.

  Serial.begin(115200);
  /* Let addressing procedure packets to be received by the receiver function
     to ease debugging or analysis */
  bus.debug = true;

  bus.strategy.set_pins(5,6);
  bus.set_receiver(receiver_function);
  bus.set_error(error_handler);
  bus.begin();
  timmy = millis();
};

void loop() {
  
  if(millis() - timmy > 5000) {
    Particle.publish("print", "List of slaves known by master: ");
    for(uint8_t i = 0; i < PJON_MAX_DEVICES; i++) {
      if(bus.ids[i].state) {
        Serial.print(" - Device id: ");
        Serial.print(i + 1); // Shifted by one to avoid PJON_BROADCAST
        Serial.print(" Device address: ");
        Serial.print(bus.ids[i].device_address[0]);
        Serial.print(".");
        Serial.print(bus.ids[i].device_address[1]);
        Serial.print(".");
        Serial.print(bus.ids[i].device_address[2]);
        Serial.print(".");
        Serial.print(bus.ids[i].device_address[3]);
        Serial.print(".");
        Serial.print(bus.ids[i].device_address[4]);
        Serial.println();
      }
    }
    Serial.println();
    Serial.flush();
    timmy = millis();
  }
  bus.receive(5000);
  bus.update();
};

The Error I am getting is: pjontest.ino:8:65: 'PJON_Packet_Info' does not name a type

Seserie, 2019-06-11:
My goal with this implementation is to make a device to monitor my Saltwater Fish tank. I am so busy with school I want to make something to monitor my tank when I am away. The particle photon would be communicating with my Home Server which would be accessible through my phone.

gioblu, 2019-06-11:
nice project @Seserie ! Why not to use just the PJON class instead of PJONMaster/Slave ?

gioblu, 2019-06-11:
@Seserie you could just program in the photon's code the master's behavior you need using just the PJON class (periodically read sensors implementing a request-response exchange, periodically actuate filter or lighting sending a request) and use static device ids (set at compilation in their program) in the other devices.

gioblu, 2019-06-11:
If you need an example for the sensor reading see [here](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/SendArbitraryValues)
If you need help for a request response exchange see [here](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/SendAndReceive)

Seserie, 2019-06-12:
Cause I guess I like to overthink things lol...

Well After a bit of research I found out that Particle has a bit of an issue with typedefs ... great. So I guess I have to turn off the preprocessor...