gioblu/PJON #305: SoftwareBitBang ESP - STM32 Connection Lost

osman-aktepe, 2019-07-04:
Hi @gioblu 

First of all, Thanks for your effort. At second, I need to tell you something, for ESP8266, the value of PJON_ADDRESSING_TIMEOUT is too high, it causes wdt reset. I used it as 1.9 seconds, I wanted you to know.

Here is my problem: I have a SWBB bus between several STM32 (slave) and ESP8266. Normally If my slave can not have an address every 30 seconds I send hearth beat from master, but sometimes communication can not be achieved.

The problem is I changed my code a little bit and now I can have no communication. What could be the cause? Do you have any idea? And how can i increase the quality of communication?

Regards

gioblu, 2019-07-04:
ciao @osman-aktepe, how much time you dedicate to polling, or better in both sketches what is the value you pass to `bus.receive`?

gioblu, 2019-07-04:
another question, how much time `iwdg_feed` lasts?

gioblu, 2019-07-04:
@osman-aktepe try to run the networkAnalysis example and please post the results here.

osman-aktepe, 2019-07-04:
Hi @gioblu 

Thanks for your quick reply.

TBH I used various values to use in bus.receive but the smallest one is 1000, iwdg_feed takes 3 micros at most.

Do you advice any values to poll and what is the best way to acquire id?

To run networkAnalysis, I need time, but I run before and result was just one drop as I remember. Now it can connect but it is not stable. 

For example sometimes it can not get any id but it can receive data from esp.

osman-aktepe, 2019-07-04:
Also I wrote a code in the loop and checked the acquired id, if it is 255 i called bus.acquire_id_master_slave(). The thing is it always returned true but the id was 255 still.

gioblu, 2019-07-04:
@osman-aktepe I suspect you are using the old version of the dynamic addressing which is deprecated. The new version is here in master, still not released, it works much better and it is more stable.

gioblu, 2019-07-04:
@osman-aktepe see https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Network/SoftwareBitBang/MasterSlaveAutoAddressing

osman-aktepe, 2019-07-04:
@gioblu probably, i am, I updated to last release but not from master but I can, but the thing is, you are not using bus id but directly giving device id, I want auto-addressing. Did I get something wrong?

gioblu, 2019-07-04:
@osman-aktepe if you want to use this in a local network you can generate randomly the 5 bytes device address in each device and pass it to PJONSlave

osman-aktepe, 2019-07-04:
What if they overlap? :) Could it break?

gioblu, 2019-07-04:
@osman-aktepe here: https://github.com/gioblu/PJON/blob/master/src/PJONMaster.h#L271 it does not handle device address duplication but it may be quickly modified to conditionally do so, it effectively could be nice for it to have a local mode. If you need those changes I will work them out in this week end.

osman-aktepe, 2019-07-04:
@gioblu Can you explain what is the difference between them because I think that they can not sync which is the problem. 
Maybe I should work on timings of SWBB, what do you think? Could it be the real problem?

 If you guide me about it, I can implement it too.

gioblu, 2019-07-04:
@osman-aktepe  Are you using the bluepill? If so I can test it for you I think I should have a couple.
The first thing to do is to run a NetworkAnalysis example and see results.
Generally if works but performance is not perfect just tweaking the `SWBB_READ_DELAY` can fix the issue. 

It would be great to have PJONSlave and PJONMaster be able to support also local networking and randomly-generated device addresses. This would cover your requirements while still supporting more complex use cases.

If there is no need to be able to contact one slave in the local bus from a foreign shared bus, this should be more than enough.

osman-aktepe, 2019-07-04:
Yeah I am using bluepill and esp-01, with 2 pins. 

In my case, just broadcasting and sending messages to the master.

If you need any help, just tell me, during that time if you think something to tweak or analyzing, I can do it now and provide what you need.

Thanks for your help.

gioblu, 2019-07-04:
Sorry @osman-aktepe the big difference is that the master-slave communication occurs in shared mode. For sure it would be more efficient in a use case like yours to have local networking support and handle randomly generated device addresses duplications.

gioblu, 2019-07-04:
I have a bluepill and a nodeMCU 1.0 is that fine?

osman-aktepe, 2019-07-04:
I did not understand the difference, is there any document or article to read?

Yeah they looks fine in esp I use pin 1,3 and for bluepill you can choose any pins with adc support.

Should I just wait for you?

gioblu, 2019-07-04:
Here is the new spec https://github.com/gioblu/PJON/blob/master/specification/PJON-dynamic-addressing-specification-v3.0.md

I think I can provide with test results tonight.

osman-aktepe, 2019-07-04:
By the way, I became obsessed with the problem:

```
if (millis() - acquireMillis > 10000) {
    Serial.print("Acquired device id: ");
    Serial.println(bus.device_id());
    acquireMillis = millis();
  }

  bus.receive(40000);
  bus.update();
  bus.receive(10000);
  
  
  if (bus.device_id() == 255 && millis() % 10 == 0) {
    delayMicroseconds(7);
    bus.acquire_id_master_slave();
  }

  readSerial();
  
  //handleSensors();

  iwdg_feed();
```

readSerial takes 0 millis, I measure, I do not touch the system, I just reset, If it connects everything works perfectly, if not, It can not obtain id too just ones it could achieve.

Is there any chance that I am looking at right point, maybe master is the problem. Can you give any idea about that too?

gioblu, 2019-07-04:
@osman-aktepe for sure this is wrong:
```cpp
  if (bus.device_id() == 255 && millis() % 10 == 0) {
    delayMicroseconds(7);
    bus.acquire_id_master_slave();
  }
```
Doing so you call `bus.acquire_id_master_slave();` every 10 milliseconds that is a tempest of requests :) that may inhibit the master to effectively respond. Save the value of millis and try to acquire an id with an interval equal to the addressing timeout you have configured.

osman-aktepe, 2019-07-04:
Noo it does not request every 10 millis but if it is dividable by 10. Basically it requests every 3 minutes.

gioblu, 2019-07-04:
That stricly depends on the time dedicated to the rest of the code and by the granularity of the increment, not sure if that is the right way to obtain the result.

osman-aktepe, 2019-07-04:
@gioblu I also wanted to ask something, 

I changed timing values and for a case i found a very good timing values, I can acquire id and receive data, after that in loop i just added one line, Serial.available(), when I add that, I can not obtain any communication. This line's duration is less than 1 ms.

Is this normal?

gioblu, 2019-07-05:
Ciao @osman-aktepe that is definitely not normal, sadly I do not have huge experience with the STM32. I have bought some bluepill from ebay to verify the port you have proposed long ago. I am now not able to obtain communication with the NodeMCU, tried pins 12, A0 and A1, and D1 on ESP8266. I suspect the problem may be related to the oscillator. On my board I do read 8MHz, that may be not quick enough to bitbang PJDL and execute the rest of the code (there is no 8MHz device which supports PJDL MODE 1).

osman-aktepe, 2019-07-05:
Hi @gioblu, Thanks for your response, I am quite experienced with STM32, if you need anything you can just ask me.
If you can install software but can not communicate, you can change timings to 44 112 52 -10 for STM32.

Can you describe your problem?

gioblu, 2019-07-05:
CIao @osman-aktepe thank you for your support, let me try again with the timing you suggest.
One question, what is the definied constant that identifies the bluepill? So I can conditionally define!
Thank you again.

osman-aktepe, 2019-07-05:
@gioblu TBH, normally i do not work with bluepill but a self designed card. The thing is my controller is STM32F103C too and your definition is quite ok, it is working. If you have problem, I have blue pill too, I can check.

gioblu, 2019-07-05:
We now have this in master:
```cpp
/* STM32F1 ---------------------------------------------------------------- */
#if defined(__STM32F1__)
  #if SWBB_MODE == 1
    #if F_CPU == 72000000L
    /* Added by github user osman-aktepe - 22/06/2018 */
      #define SWBB_BIT_WIDTH   44
      #define SWBB_BIT_SPACER 112
      #define SWBB_ACCEPTANCE  40
      #define SWBB_READ_DELAY -10
    #endif
  #endif
#endif
```

although If I use the Arduino IDE select tools->board->Generic STM32F103C series
and I flash the thing, the constant `__STM32F1__` is not defined.

osman-aktepe, 2019-07-05:
How do you flash? Do you use bootloader?

gioblu, 2019-07-05:
I have both jumpers set to 0 and I have just flashed it via usb, I do see the new code being executed so I suspect the flashing procedure goes through.

osman-aktepe, 2019-07-05:
I mean, do you have st-flash or do you have stm32bootloader like this:
https://github.com/rogerclarkmelbourne/STM32duino-bootloader

osman-aktepe, 2019-07-05:
But installing bootloader for bluepill is a little bit problematic, so i recommend you to solve it by defining them by yourself bypassing __STM32F1__ and define those values as default :)

gioblu, 2019-07-05:
Thank you @osman-aktepe I will buy also a couple of Maple mini with pre-installed bootloader.

osman-aktepe, 2019-07-05:
By the way, i will try your new implementation, because my case a little bit urgent i need to solve it until next wednesday, i can use chips unique ids to address. But i think that the problem i face about swbb configuration not your old implementation.

If you need any data or case, i can provide you.

gioblu, 2019-07-05:
@osman-aktepe send me the code if you want: gioscarab@gmail.com I will review it for you

osman-aktepe, 2019-07-05:
I can send my code but it is hard for you to make it work because it uses specific sensors and libraries but you can review it. I am sending right them now.

osman-aktepe, 2019-07-11:
This was caused by Serial comm. If you wanna use debuggin you should be careful because it hurts SWBB. 

Thanks for your support @gioblu

closed 2019-07-11 by osman-aktepe

gioblu, 2019-07-11:
Ciao @osman-aktepe thank you very much for reporting back results :)