gioblu/PJON #309: Support for 328PB MCU

esaheiskanen, 2019-10-17:
Hi!

I was planning to test PJON with uStepper. It has 328PB MCU, but it failed to compile. I checked out source and it seems that 328PB is not supported. How difficult it is to add support for it?

gioblu, 2019-10-17:
Ciao @esaheiskanen thank you for your report. Reading this document:
https://www.pololu.com/file/0J1464/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf
seems it should be quite straight-forward.

Could you share the compilation error you obtained?

esaheiskanen, 2019-10-17:
I was just reading the same document!

This is only a clip of the compile errors. It seems that most of the errors are caused by missind defines. 

`/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/strategies/SoftwareBitBang/SoftwareBitBang.h:277:7: note: in expansion of macro 'PJON_IO_PULL_DOWN'
       PJON_IO_PULL_DOWN(_input_pin);
       ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:277:42: error: 'PJON_IO_PIN_TO_BIT' was not declared in this scope
       bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); \
                                          ^
/Users/esaheiskanen/.platformio/packages/framework-arduinoavr/cores/MiniCore/Arduino.h:132:49: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                                                 ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:277:7: note: in expansion of macro 'bitWrite'
       bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); \
       ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:294:9: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
         PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_DDR_REG(P), P, V) \
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:310:9: note: in expansion of macro 'PJON_IO_MODE'
         PJON_IO_MODE(P, INPUT); \
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/strategies/SoftwareBitBang/SoftwareBitBang.h:277:7: note: in expansion of macro 'PJON_IO_PULL_DOWN'
       PJON_IO_PULL_DOWN(_input_pin);
       ^
In file included from /Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_ARDUINO_Interface.h:35:0,
                 from /Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/PJON_Interfaces.h:4,
                 from /Users/esaheiskanen/.platformio/lib/PJON_ID245/src/PJON.h:94,
                 from src/main.cpp:9:
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:285:68: error: 'PJON_IO_PIN_TO_PORT_REG' was not declared in this scope
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
                                                                    ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:273:15: note: in definition of macro 'PJON_IO_ATOMIC_WRITE'
     if ((int)(A) < 0x40) { bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); }  \
               ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:309:9: note: in expansion of macro 'PJON_IO_WRITE'
         PJON_IO_WRITE(P, LOW); \
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/strategies/SoftwareBitBang/SoftwareBitBang.h:279:9: note: in expansion of macro 'PJON_IO_PULL_DOWN'
         PJON_IO_PULL_DOWN(_output_pin);
         ^
In file included from src/main.cpp:1:0:
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:273:63: error: 'PJON_IO_PIN_TO_BIT' was not declared in this scope
     if ((int)(A) < 0x40) { bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); }  \
                                                               ^
/Users/esaheiskanen/.platformio/packages/framework-arduinoavr/cores/MiniCore/Arduino.h:132:49: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                                                 ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:273:28: note: in expansion of macro 'bitWrite'
     if ((int)(A) < 0x40) { bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); }  \
                            ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:285:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
           ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:309:9: note: in expansion of macro 'PJON_IO_WRITE'
         PJON_IO_WRITE(P, LOW); \
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/strategies/SoftwareBitBang/SoftwareBitBang.h:279:9: note: in expansion of macro 'PJON_IO_PULL_DOWN'
         PJON_IO_PULL_DOWN(_output_pin);
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:277:42: error: 'PJON_IO_PIN_TO_BIT' was not declared in this scope
       bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); \
                                          ^
/Users/esaheiskanen/.platformio/packages/framework-arduinoavr/cores/MiniCore/Arduino.h:132:49: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                                                 ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:277:7: note: in expansion of macro 'bitWrite'
       bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); \
       ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:285:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
[... 661 lines cut ...]
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:273:28: note: in expansion of macro 'bitWrite'
     if ((int)(A) < 0x40) { bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); }  \
                            ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:294:9: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
[... cut ...]
       bitWrite(*(A), PJON_IO_PIN_TO_BIT(P), V); \
       ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/interfaces/ARDUINO/PJON_IO.h:294:9: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
         PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_DDR_REG(P), P, V) \
         ^
/Users/esaheiskanen/.platformio/lib/PJON_ID245/src/strategies/ThroughSerial/ThroughSerial.h:272:7: note: in expansion of macro 'PJON_IO_MODE'
       PJON_IO_MODE(_enable_RS485_txe_pin, OUTPUT);
       ^
*** [.pio/build/ATmega328PB/src/main.cpp.o] Error 1`

gioblu, 2019-10-17:
yes, thank you for sharing, so change this line: https://github.com/gioblu/PJON/blob/master/src/interfaces/ARDUINO/PJON_IO.h#L98
in:
```cpp
defined(__AVR_ATmega328__) || defined(__AVR_ATmega328P__) ||  defined(__AVR_ATmega328PB__)
```
And change this line:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L27
in:
```cpp
  defined(__AVR_ATmega328__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__)
```
After those changes, I think it should work out of the box if your hardware uses an external 16MHz oscillator and the chips are effectively executing the code equally and at the same speed.

gioblu, 2019-10-17:
@esaheiskanen it may have sense to define `PJON_INCLUDE_SWBB` to avoid the inclusion of any other  strategy out of [SoftwareBitBang](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) and so avoid other potential errors related to unsupported strategies.

esaheiskanen, 2019-10-17:
Thanx! now it compiles. I'll check if it works correctly otherwise.

esaheiskanen, 2019-10-17:
I'm using ThroughSerial and rs485.

gioblu, 2019-10-17:
Sorry @esaheiskanen I took for granted you were interested in SoftwareBitBang.
Thank you for your feedback, let me know how it goes :)

esaheiskanen, 2019-10-17:
Can confirm that it works - at least for me with ThroughSerial and rs485 :)

Do I need to do pull requestor will you add this?

gioblu, 2019-10-18:
Ciao @esaheiskanen thank you very much.
Soon the version 12.0 will be released, after that I will push this change. 

Thank you again! :)

closed 2019-10-20 by esaheiskanen