gioblu/PJON #311: How to recover from PJON_CONNECTION_LOST

notreallybob, 2019-10-23:
Hi,

I've been playing with SoftwareBitBang between 2 esp8266s with good success.
When I disconnect the slave, (eg in the SendAndReceive example), after a certain timeframe (around 5s) the expected error code of PJON_CONNECTION_LOST shows up on the master. If I reconnect the slave, the master still does not recover/continue sending.

My question is: Does PJON automatically recover, or is there a way to make PJON recover. Or do we have to reset/reboot each device once communication is lost?

gioblu, 2019-10-23:
Ciao @notreallybob what you describe is not an issue of PJON, but just a limitation of the SendAndReceive example. If one of the two devices is unreacheable, the chain of packets is interrupted, and one of the 2 sketches must be reset to start back the ping-pong.

As you can se from the code here:
https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/SendAndReceive/Device1/Device1.ino#L29
the Device1 tries to send once when powered up, but it does not check the result, so, if the Device2 is not present for more than 4 seconds (the maximum timeout for a packet transmission over SoftwareBitBang), the sketch will just wait for a packet from Device2.

gioblu, 2019-10-24:
Ciao @notreallybob I have run some tests on other systems and I am not able to replicate the issue, so I think it must be ESP related. I will now start ESP testing. One thing I do see is that you schedule a packet sending every second, also if the earlier were not successful and may still be in the buffer waiting to be delivered or to fail. Did you get any `PJON_PACKETS_BUFFER_FULL` error? What happens changing this line in the master:
```
if (millis() - lastsend >= 1000) {
```

in 
```
if ((millis() - lastsend >= 1000) && !bus.update()) {
```
`bus.update()` returns the number of packets in the buffer.

notreallybob, 2019-10-25:
This is embarrassing. I can no longer replicate the error. I was able to replicate the error fairly consistently the last few days, but not today. Nothing should have changed. Last night I just unplugged usb on both ESPs after sending the email with code, and today reconnect usb, but cannot reproduce the error again after lengthy disconnect/reconnect routine.

I suppose we just leave this "issue" here and not worry further until or unless I can reproduce the problem again.

gioblu, 2019-10-25:
Ciao @notreallybob no problem :)
Consider that ESP8266 has been proven vulnerable by @Matheus-Garbelini here:
https://github.com/Matheus-Garbelini/esp32_esp8266_attacks

Be sure to have your toolchain updated.
Feel free to re-open the issue if you find a way to replicate it.

closed 2019-10-25 by gioblu

hal295, 2021-05-31:
Hi, I think I have similar problem with ESP8266 (NodeMcu) and I am really lost.
I am using two ESP8266 (D1 mini) with SoftwareBitBang strategy.
Sending data every 3 seconds from device 1 to device 2 and from device 2 to device 1.
Its working nicely, but after some time connection with device is lost and cannot be restored. 
Restarting only one device does not help. I have to turn off the power to both devices.
I am not using replies, only bus.send(). I tried even send_packet();

Loop looks like this:
void loop() {
  bus.receive(20000);
  bus.update();
}

Code in setup:
bus.strategy.set_pin(5);
bus.set_error(error_handler);
bus.set_receiver(receiver_function);
bus.begin();

I tried many things but without success. Can you please help me?
Is there some way to restore communication? Clear buffer, reinitialisation? 
Thank you.