gioblu/PJON #314: ATMEGA328PB compatibilty

robkettridge, 2019-11-05:
Hello

I have an ongoing project that is at the production board stage and so far have been populating the units with ATMEGA328P processors. However I've just accidentally ordered a batch of the PB models which claim to be backwards compatible, but when I try a board with this processor it all seems to work, apart from one digital output that only seems to work if I use it as a PWM output, and the PJON communication doesn't seem to work. I've just tried updating my library to the latest here as mentions adding 328PB compatibility but to no avail. I'm using software bitbang on pin 12 between two ATMEGA chips and SWBB_MODE 3. Any suggestions? 

Thanks

gioblu, 2019-11-05:
Ciao @robkettridge there may be some subtle differences in the calls durations.
Try to change `SWBB_READ_DELAY` here: https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L53
on both sides and see if works. If you don't have any success it may mean that the ATmega328PB is a little slower than the 328P and so it is not able to run mode 3, in that case, mode 2 is the only answer.

gioblu, 2019-11-05:
I still did not have the time to acquire one and test SoftwareBitBang with it but I have ordered now some :)

robkettridge, 2019-11-05:
Just tried changing the SWBB mode. Mode 3 (as it was) doesn't show any signs of data, mode 2 occasionally flashes up showing received data, mode 1 comes and goes a few times a second. Unfortunately I don't really have the ability to debug this properly as all my breadboarding versions are the older P variants which I developed these boards on. So I only have the valid data lights on each end of the link to show when I have data. This switches on on the receiving end when receiver_function() is called.

gioblu, 2019-11-05:
@robkettridge there must be some difference. Try fiddling with `SWBB_READ_DELAY` probably with a slightly higher or lower value will work (who knows maybe the PB is quicker?), make a trial and error session flashing both devices with a new value and see results.

gioblu, 2019-11-05:
@robkettridge obviously the lower is the bandwidth the easier is to find the right timing, so may have sense to start from mode 1.

gioblu, 2019-11-05:
Another thing @robkettridge , it may help to run the NetworkAnalysis example to debug performance while testing new values of `SWBB_READ_DELAY`

gioblu, 2019-11-10:
Hei @robkettridge have you managed to find the correct setup?

robkettridge, 2019-11-10:
Unfortunately not. As I said I'm not really in the position to be doing evaluation with the different. I've ended up ordering some of the original parts as I'm on a tight schedule to get these other units working. Once I've got those out the door I might get some time to go back to prototyping and see if I can get it working.

gioblu, 2019-11-10:
@robkettridge I hope I will get the ICs as soon as possible to get some testing done.

willie68, 2020-01-31:
For my model railway I'm designing a new decoder for turnouts, lights, and other parts. And i want to use PJON as the main protocol (because DCC ist a little bit more foccussed on digital train management, but i just want to control the enviroment without a PC):
Therefor i have a newly designed board and i need some more pins, so i just used the ATMega328PB. If i try to compile the sources, i get an error of undefined `PJON_IO_PIN_TO_PORT_REG`, because the mcu is not listed in PJON_IO.h. (And in the timing it's missing, too) 
I'm using the minicore for hardware setup. 
Since the PB is like a normal 328, I simply added `|| defined(__AVR_ATmega328PB__)` to the section, where i find the `|| defined(__AVR_ATmega328P__)` 
Now it's compiling without any error. 
Next i will see, if a normal Arduino Mega can talk to this decoder...

gioblu, 2020-01-31:
Ciao @willie68 thank you for your report. Let me know how it goes :)

robkettridge, 2020-01-31:
I could get it to compile but couldn’t get a PB unit to talk to or listen to the bus.

gioblu, 2020-02-01:
Ciao @willie68 I am really curious to see the application running :)

Here:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h
I suggest you to try on the PB to include the timing constants of the 328 and see what happens, as you did in PJON_IO.h

If that does not work I suggest you to:
- Try to fiddle with the value of `SWBB_READ_DELAY` on PB side while running the networkAnalsysis example and see if you can get better results
- if nominal functionality cannot be achieved only changing the bit reading offset or `SWBB_READ_DELAY` that may mean that `SWBB_BIT_WIDTH` and `SWBB_BIT_SPACER` timing constants may need to be tweaked as well, as it is done in other targets. In this case I generally use the oscilloscope to measure bit durations and tune the constants to obtain the specified duration.

willie68, 2020-02-01:
Just fiddling a little around with SWBB 2, this will be my best parameter...
      #define SWBB_BIT_WIDTH   33
      #define SWBB_BIT_SPACER  88
      #define SWBB_ACCEPTANCE  56
      #define SWBB_READ_DELAY   8

Results from networktests:
17:40:15.383 -> Packet Overhead: 10B - Total: 640B
17:40:15.383 -> Bandwidth: 1920B/s
17:40:15.383 -> Data throughput: 1280B/s
17:40:15.383 -> Packets sent: 64
17:40:15.383 -> Mistakes (error found with CRC): 0
17:40:15.383 -> Fail (no acknowledge from receiver): 0
17:40:15.383 -> Busy (Channel is busy or affected by interference): 0
17:40:15.383 -> Accuracy: 100.00 %

gioblu, 2020-02-02:
ciao @robkettridge thanks to @willie68 we have ATmega328PB compatibility for MODE 1 and 2 on pin 10, as soon as I will receive the hardware I will try extend compatibility to other pins and to mode 3 and 4.

closed 2020-02-02 by gioblu