gioblu/PJON #317: Teensy ThroughSerial RS-485 Blink example not working

garageboxnv, 2019-12-05:
Hi,

For a home project I am trying to deploy PJON as a RS485 bus.
To start I loaded the Blink RS485 example and loaded the transmitter and a receiver to a teensy LC.
Using pin 17 to control the RTS of the rs485.

I see that the transmitter is sending data but goes into error mode because of connection lost.
I also see that the receiver does not call the receiver function to start.

**Transmitter**
`
/* PJON MAX485 serial modules RS485 communication example */
#define HWSERIAL Serial1
#include <PJON.h>

// <Strategy name> bus(selected device id)
PJON<ThroughSerial> bus(45);

void setup() {
  // Initialize LED 13 to be off
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  // Initialize Serial instance used for PJON communication
  Serial1.begin(9600);
  Serial.begin(9600);
  bus.strategy.set_serial(&Serial1);
  // Set enable pins
  bus.strategy.set_enable_RS485_pin(17);
  bus.strategy.set_RS485_delay(1);
  bus.set_error(error_handler);
  bus.begin();
  // Send B to device 44 every 1.5s
  bus.send_repeatedly(44, "B", 1, 1500000);
}

void error_handler(uint8_t code, uint16_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST)
    digitalWrite(LED_BUILTIN, HIGH); // Light up LED 13 if a packet transmission failed
}

void loop() {
    bus.update();
    bus.receive();
};`

**receiver**

/* PJON MAX485 serial modules RS485 communication example */
#define HWSERIAL Serial1
#include <PJON.h>

// <Strategy name> bus(selected device id)
PJON<ThroughSerial> bus(44);

void setup() {
  // Initialize LED 13 to be off
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  // Initialize Serial instance used for PJON communication
  Serial1.begin(9600);
  bus.strategy.set_serial(&Serial1);
  // Set enable pins
  bus.strategy.set_enable_RS485_pin(17);
  bus.begin();
  bus.set_receiver(receiver_function);
};

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */
  //if(payload[0] == 'B') {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(3);
    digitalWrite(LED_BUILTIN, LOW);
    delay(3);
  //}
}

void loop() {
  bus.update();
  bus.receive();
};

gioblu, 2019-12-05:
Ciao @garageboxnv I suggest you to see if the example works using bare serial connection (remove RS485 in the middle). If it works then try to set a higher delay in between the enable pin state change and the effective use of the line: https://github.com/gioblu/PJON/blob/master/src/strategies/ThroughSerial/ThroughSerial.h#L279

That may solve the issue.

garageboxnv, 2020-01-21:
Thanks, this helped me debug, some connection was wrong on the RS485 board

closed 2020-01-21 by garageboxnv