gioblu/PJON #319: Mistakes in book

drtrigon, 2019-12-17:
I think page 93 (DualUDP) of the book has a mistake at the end of the first paragraph. The last sentence should read: "Like the **LocalUDP** strategy ..." or am I wrong?

gioblu, 2019-12-27:
Yes, thank you very much @drtrigon :)
I hope the book will be useful for your research.

closed 2020-07-10 by gioblu

drtrigon, 2020-11-05:
Is you offer to sell a signed book to me still present?

Is it possible to also get a digital copy of your book?

gioblu, 2020-11-05:
Ciao @drtrigon happy to hear you, sure. Sadly we are still working on the digital version, we hope it will be ready soon.

drtrigon, 2020-11-06:
How did you deliver the book to the printing service? pdf, ps or dvi? Can I get a copy of this file together with the book?

Also; do you need help on reviews or else?

gioblu, 2020-11-07:
Ciao @drtrigon sorry for the late answer. The production files do not contain links and proper interactivity an e-book should have but we are working on it. I would be happy to send you the new version of the book, updated to the new coming v13.0. Wait to purchase until the new release is out and the new books will be available in the shop.

drtrigon, 2020-11-09:
> Ciao @drtrigon sorry for the late answer. The production files do not contain links and proper interactivity an e-book should

Actually that would be ok for me, links are nice but not a must have.

I take v12 or v13 of the printed book I just would like to make sure that:
- you sign it
- my name is mentioned in the book (as it is in v12 ;)