gioblu/PJON #320: Feature Request: Implement probe request protocol as a stratagy

happytm, 2019-12-23:
Thank you for your great out of box software. I have been working on simple IOT sensor network solely using probe request protocol. It is very power efficient for battery powered devices. Please implement this protocol in your software. There are 2 sketches for my software called ProbeReceiver.ino which is single main gateway on the network and ProbeSender.ino is used by all battery powered ESP devices on network. Link to my code is below:

https://github.com/happytm/BatteryNode

Thanks.

happytm, 2020-05-05:
Can some one take a fresh look at this request please?

Thanks.

gioblu, 2020-05-05:
Ciao @happytm sorry for the late response, we are hardly working on breaking changes scheduled long ago. Do you have a link to the documentation or the specification of the protocol?

happytm, 2020-05-05:
Other than information at link in first post there is no documentation because it is very simple code of 300 lines if you look at ProbeReceiver.ino sketch. Basically the gateway receive all data from remote battery powered sensors via Probe Request and automatically convert it to json format and built in MQTT broker publishes these data so any MQTT client can connect to the gateway and receive data from whole network of devices. All the devices are ESP8266. Gateway can also receives MQTT messages from any MQTT client and send it to remote devices to control them. I use android app call MQTT Dash https://play.google.com/store/apps/details?id=net.routix.mqttdash&hl=en_US to access the gateway.

Thanks.