gioblu/PJON #321: Udp and SoftwareBitBang

mschmidi, 2019-12-27:
Dear guys,

First if all happy holliday to you all.
Now to my question. I’ve been playing with SoftwareBitBang a couple of months ago, and with the support of the community i was able to make it work between 2 arduinos and 2 wemos d1mini.
Unfortunately due to lack of time and success, i left this topic sitting on my work bench since.
Now i would like to restart a and give it another try.
Before i do so i tought to share my final goal with you, so you comment on paths to follow and so on.

I would like to spread out many small arduinos in my garden manly hooked up to a relay to switch lights and sprinkler heads all of the connected thru an coax cable ending up on a wemos mini which i plan to use as a gateway between standart UPD protocol used by my home automation and the SoftwareBitBang for the arduino switches. 

I would like to use an wemos as gateway as i like their possibilities like OTA an so on. However if i need to use and arduino with an eth shield i would do so. 

No to my tests so far. As i wrote earlier i have been able to communicate between to arduinos or 2 Wemos using SoftwareBitBang. However as soon i try to mix them up i was not able to make them work. (If i remember right someone tried to explain me why but i dit not get it 😳) The i tried using an eth shield on a arduino nano (which worked for the BitBang) sadly i got conflicts between the eth and the pjon library. This was the stage i gave up since my basic skills don’t allow me to change libraries on my on...

So i’m wondering am I the only one trying such a scenario? Or did i miss a key Element of Pjon protocol? 

Thank you so much for your comments.
Br M.

fredilarsen, 2019-12-27:
@mschmidi I have in my automation setup two SoftwareBitBang local buses in different buildings. To avoid pulling still another wire between the buildings I use one Arduino Nano with an Ethernet card on each bus to connect the 2 physical buses into one "virtual" local bus through my wired Ethernet LAN.

The Nano "switches" use the [BlinkingRGBSwitch](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingRGBSwitch) example (with some additions to report traffic statistics in my [ModuleInterface](https://github.com/fredilarsen/ModuleInterface) setup). I also changed it to use DualUDP instead of LocalUDP, to avoid packets being broadcast on all my WiFi access points unnecessarily.

I think the only problems I had was the RAM amount, so as you can see in the example the send queue size has been reduced with the PJON_MAX_PACKETS preprocessor definition.

gioblu, 2019-12-27:
Ciao @mschmidi happy to hear from you again :)

If you are making the first steps with PJON I would suggest you to avoid to run a mixed voltage level bus (3v3 and 5v) because that adds various potential issues (and also the ESP is not that secure specially for a long term deployment). For entry level use / evaluation I would suggest you to start with Arduino nanos for slaves/nodes and arduino nano, uno or mega + ethernet shield if you have the hardware available. 

When building the SoftwareBitBang side remember to use the pull-down and current-limiting resistors as suggested here https://github.com/gioblu/PJON/wiki/Mitigate-interference and here 
https://github.com/gioblu/PJON/wiki/Protective-circuitry 

Also be sure to choose pins which work with SoftwareBitBang, see here https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang#compatibility

Could you paste here the errors you got when trying to compile using the Eth shield?

If you need direct support, you can also join our gitter chat here: https://gitter.im/gioblu/PJON

mschmidi, 2019-12-27:
Hi guys,
As usual, this was quick! 
I will try find some time within the next few days and comment my results. Thank you all for your support!

closed 2020-01-28 by gioblu