gioblu/PJON #322: ESPNOW Examples runtime failure using esp_wifi_set_channel

justoke, 2019-12-28:
I have just tried the Arduino [example ](https://github.com/gioblu/PJON/tree/master/examples/ESP32/ESPNOW/PingPongArduinoIDE)for the ESPNOW strategy and encountered a run-time error. 

```
Transmitter started.
E (234) event: Event loop not initialized via esp_event_loop_init, but esp_event_send called
E (235) wifi: esp_wifi_set_channel 1453 wifi not start or sniffer not enabled, should start wifi or enable sniffer first
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x4008cf5c
file: ".pio\libdeps\esp32dev\PJON_ID245\src/strategies/ESPNOW/../../interfaces/ARDUINO/ESPNOWHelper.h" line 179
func: bool ENHelper::begin(uint8_t, uint8_t*)
expression: esp_wifi_set_channel(_channel, WIFI_SECOND_CHAN_NONE)
```

It is exactly as described in this [espressif-idf issue 4311](https://github.com/espressif/esp-idf/issues/4311) .  It appears to be a change to ESPNOW implementation and not a bug as such. I have posted the code and asked what the suggested implementation should be. In the meantime I will try to fix it and report back here.

gioblu, 2019-12-28:
Ciao @justoke thank you very much for your support. 
@xlfe do you have any idea why this issue may occur?

justoke, 2019-12-28:
I have found the issue and will raise a PR and an example using ESPNOW with PlatformIO in VSCode

gioblu, 2019-12-28:
Ciao @justoke I have answered to your email request, thank you for your support :)

gioblu, 2019-12-29:
Ciao @justoke thank you very much, I have accepted your pull-request, will follow minor code-style changes.

closed 2019-12-29 by gioblu