gioblu/PJON #324: MQTTTranslate and ThroughSerial strategy router/switch

garageboxnv, 2020-01-21:
I am creating a PJON network to interact with relays and input boards, which I want to sync with a router to a MQTT server.
I was planning to make 1 master board with a translation table between PJON nodes and MQTT nodes, however due to the new MQTTTranslate strategy, this seems not the way to go.

Unfortunately when creating the router, I get following error:
[logMQTT_Serial_Router.log](https://github.com/gioblu/PJON/files/4093110/logMQTT_Serial_Router.log)

`#define HWSERIAL Serial1
#define MAADDR 1
#define DEBUG false

#define PJON_INCLUDE_ETCP
#define PJON_INCLUDE_MQTT
#define PJON_INCLUDE_TS
#include <PJON.h>
#include <PJONRouter.h>

StrategyLink<ThroughSerial> serial;
StrategyLink<MQTTTranslate> mqtt;

PJONAny serialBus(&serial, (uint8_t[4]){0, 0, 0, 1});
PJONAny mqttBus(&mqtt, (uint8_t[4]){0, 0, 0, 2});`

Is this not yet implemented, or did I misunderstood the documentation?

fredilarsen, 2020-01-28:
@garageboxnv The routing is based on several buses with different devices talking to each other through a router. In your setup I guess that one of the MIRROR modes would be the best choice, and this does not fit directly into the routing functionality, though we will evaluate if it can be.

The MQTT strategy can run in BUS or MIRROR modes. The BUS modes allow one device to talk to another device so that they can form a bus _through_ the MQTT broker. The MIRROR modes use a more common approach for MQTT devices, namely only publish outputs and subscribe to inputs within its own MQTT topic, depending on some other MQTT based system to inspect and control the devices.

If you look at the [Gateway](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/MQTTTranslate/SWBB-MQTT-Gateway/Gateway/Gateway.ino) example and its companion [SomeDevice](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/MQTTTranslate/SWBB-MQTT-Gateway/SomeDevice), this includes routing-like functionality that forwards every incoming packet to the MQTT broker as an output, and forwards inputs to the devices on the bus.
README  for the Gateway setup is still not finished.

This allows one gateway to mirror all devices on a bus of any strategy, to individual topics in the MQTT broker.
This is be equivalent to using the MQTT strategy with the same MIRROR mode in every device and let them talk directly to the broker without a gateway, but then every device requires an Ethernet/WiFi connection. But with a gateway only one device needs the Ethernet/WiFi capability.

Direct answer to your question: The MQTT MIRROR modes are so far not usable with the router classes. BUS modes will probably fit, but are for special cases only.

I hope this clarified some details. Some more polish and docs will come.

gioblu, 2020-01-28:
Ciao @garageboxnv, looking at the log, it seems you have not defined `PJON_INCLUDE_MQTT`, being still experimental code MQTTTranslate strategy it is not automatically included as other more mature strategies. That should fix the compilation error. I will add a remark about this in the documentation, thank you for reporting this.

gioblu, 2020-01-28:
Here docs updated, sorry for the late response: https://github.com/gioblu/PJON/commit/37304aea865748df210ceeb9a2f93884987f07fa

fredilarsen, 2020-01-29:
The MQTT feature is new in master and not present in v12.0.
To use it, make sure to download master and not a release, until v12.1 is released.

PizzaProgram, 2020-12-15:
@garageboxnv 
- Did you succeeded with your project? 
- Can you please share your sketch / code how you did it?

closed 2022-02-10 by garageboxnv