gioblu/PJON #332: Stop working when using SHTSensor

dtony, 2020-02-21:
I am trying to send a temperature data of a SHT31 on I2C from an Arduino Pro Micro to an Arduino Uno using PJON. Communication works great between the two devices but as soon is I am doing SHT.init() there is no more PJON communication. I tried SoftwareBitBang and OverSampling on several pins of the Pro Micro (4,6,8 and even 10). I use pin 12 of the UNO.

I have a 100 Ohm resistor between the two devices for testing.

Any help would be more than appreciated.
Thank you.

gioblu, 2020-02-21:
Ciao @dtony I suspect the library you use for the sensor uses interrupts, that breaks the sync of OverSampling and SoftwareBitbang. One solution may be to call `init`, make the measurement and then call `cleanup` and use PJON until the next measurement.

dtony, 2020-02-22:
Thank you, it look like that solved the issue. I was indeed calling the sht call and bus.update() each time on the loop. By calling the sht read only once every few seconds it solved the issue. I was also able to solve a similar issue I had with Ethernet and PubSub library, calling their update function only once every second and bus.update() every time solved the issue. 

Thanks.

closed 2020-02-22 by dtony

gioblu, 2020-02-22:
Ciao @dtony happy to hear that :)
Happy tinkering!