gioblu/PJON #333: [Request] Implement use of AT commands for use in radios like Reyax RYLR896 and similar

rebel2234, 2020-02-24:
Would it be possible to implement some methods that would allow communications to take place with the use of AT commands over serial, SPI, etc. etc.  In modules such as the Reyax RYLR405, RYLR406, RYLR895 and RYLR896.  Maybe the methods could be written in such a way that they are not specific to these modules but could also work with other devices that use AT commands such as RockBLOCK MK2 Iridium Satellite module.

gioblu, 2020-02-25:
Ciao @rebel2234 sorry for the delay. Yes a strategy for that hardware that makes use of AT commands would be really cool, and enlarge our operational envelop.

I am now in Milan, all is paralized because of the coronavirus outbreak, so I am not sure how much time the test units will take to reach me here.

If you are interested in trying to develop the strategy I can support you here or directly on gitter: https://gitter.im/gioblu/PJON

gioblu, 2020-04-04:
Ciao @rebel2234 arrived the modules, I will tomorrow start to work out the support. Thank you for rising this :)

rebel2234, 2020-04-20:
@gioblu How's it coming along?  Sorry I can't be of much help as I am not very good at programming.

gioblu, 2020-04-22:
Ciao @rebel2234 sorry for the late response, I am not sure why but the 2 modules I have received seems totally unresponsive, do you know if the modules need to be flashed with the firmware or there is something i am missing?

rebel2234, 2020-04-22:
They should work out of the box. You will have to use AT commands to set
parameters (bandwidth, coding rates, etc.)

AT command guide can be found here.
https://www.google.com/url?sa=t&source=web&rct=j&url=http://reyax.com/wp-content/uploads/2018/01/REYAX-Lora-AT-COMMAND-GUIDE_EN.pdf&ved=2ahUKEwjx8sTr9_zoAhXVrJ4KHUtGDZUQFjABegQIAxAC&usg=AOvVaw2mQhKeuS4PoTg3cNCiPJx7

Maybe use AT command to reset to default and try again.

Broc