gioblu/PJON #337: SWBB half duplex issue

sacreyoubeurt, 2020-03-18:
Hello,
I'm working on a project of MIDI (or equivalent) modular controller based on Arduino Uno boards.
1) Peripheral Arduino boards have to send their sensors values (buttons, potentiometers, faders) to a central Arduino board connected to the computer.
2) In return, the central board have to send to the peripheral boards the correspondant values from a software (probably Ableton Live) to insure synchronisation between sensor and software values when sensors are encoders or motorized faders.
3) Additionally, the central board would send strings from the computer to peripheral boards in order to display on OLED units.
I'm trying to use PJON with SoftwareBitBang strategy as communication protocol between the boards.

At the moment I succeed in doing 1) send sensors values from peripheral to central board.
But when I try to implement 2) I've got the following issues : 
- reception peripheral>central is really troubled (few values in central serial monitor)
- no transmission or reception central>peripheral (no values in peripheral serial monitor)
I think this is a half duplex issue, boards trying to transmit/receive at the same time, but I don't know how to figure it out.

Here is the code of the peripheral board : 
`
#define pinVal1 A0
#define pinVal2 A1
#define pinVal3 A2
#define pinVal4 A3
#define pinVal5 2
#define pinVal6 3
#define pinVal7 4
#define pinVal8 5

#define nbrIN 8

#define pinPJON 12
#define PJON_INCLUDE_SWBB
#include <PJON.h>

const int id = 2; // Peripheral bus id
int idM; // Central bus id
const char type = 'A'; // Peripheral unit type
char typeM; // Central unit type
unsigned long time;
int lenght = nbrIN+1;
uint8_t val[nbrIN];
uint8_t newval[nbrIN];

PJON<SoftwareBitBang> bus(id);

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched*/

  if(!bus.update()) {
    idM = (packet_info.sender_id) ;
    typeM = ((uint8_t)(payload[0])) ;
    newval[1] = ((uint8_t)(payload[1])) ;
    newval[2] = ((uint8_t)(payload[2])) ;
    newval[3] = ((uint8_t)(payload[3])) ;
    newval[4] = ((uint8_t)(payload[4])) ;
    newval[5] = ((uint8_t)(payload[5])) ;
    newval[6] = ((uint8_t)(payload[6])) ;
    newval[7] = ((uint8_t)(payload[7])) ;
    newval[8] = ((uint8_t)(payload[8])) ;

    Serial.print("Central unit of ID ") ;
      Serial.print(packet_info.sender_id) ;
      Serial.print(" and type ") ;
      Serial.print(typeM);
      Serial.print(" newval1: ") ;
      Serial.print(newval[1]);
      Serial.print(" newval2: ") ;
      Serial.print(newval[2]);
      Serial.print(" newval3: ") ;
      Serial.print(newval[3]);
      Serial.print(" newval4: ") ;
      Serial.print(newval[4]);
      Serial.print(" newval5: ") ;
      Serial.print(newval[5]);
      Serial.print(" newval6: ") ;
      Serial.print(newval[6]);
      Serial.print(" newval7: ") ;
      Serial.print(newval[7]);
      Serial.print(" newval8: ") ;
      Serial.print(newval[8]);
    }
}
     
    
void transmitter_function() {
  uint8_t packetA[lenght] = {
    (uint8_t)(type),
    (uint8_t)(val[1]),
    (uint8_t)(val[2]),
    (uint8_t)(val[3]),
    (uint8_t)(val[4]),
    (uint8_t)(val[5]),
    (uint8_t)(val[6]),
    (uint8_t)(val[7]),
    (uint8_t)(val[8])
    };
  bus.send_packet(idM, packetA, lenght) ;
}

void setup() {
  pinMode(pinVal5, INPUT_PULLUP);
  pinMode(pinVal6, INPUT_PULLUP);
  pinMode(pinVal7, INPUT_PULLUP);
  pinMode(pinVal8, INPUT_PULLUP);
  Serial.begin(115200);
  bus.strategy.set_pin(12);
  bus.begin();
  bus.set_receiver(receiver_function);
  bus.set_communication_mode(PJON_HALF_DUPLEX);
}

void loop() {
  
  val[1] = analogRead(pinVal1)>>3;
  val[2] = analogRead(pinVal2)>>3;
  val[3] = analogRead(pinVal3)>>3;
  val[4] = analogRead(pinVal4)>>3;
  val[5] = !digitalRead(pinVal5)*127;
  val[6] = !digitalRead(pinVal6)*127;
  val[7] = !digitalRead(pinVal7)*127;
  val[8] = !digitalRead(pinVal8)*127;
  
  bus.receive(1000);
  if(millis() - time > 10) {
  time = millis();
  transmitter_function();
}
  bus.update();
};`

And here is the code of the central board : 

`#define PINtransmitPJON 12
#define PJON_INCLUDE_SWBB
#include <PJON.h>

PJON<SoftwareBitBang> bus(1);

unsigned long time;
char type;
uint8_t count = 0;

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */
     
  if(!bus.update()) {
    type = ((uint8_t)(payload[0])) ;
    count = ((uint8_t)(payload[1])) ;
    
    Serial.print(type);
    Serial.println(count);
  }
}
      
void transmitter_function() {
  uint8_t packetM[2] = {(uint8_t)('M'), (uint8_t)(count)};
  bus.send_packet(2, packetM, 2) ;
}

void setup() {
  Serial.begin(115200);
  bus.strategy.set_pin(12);
  bus.begin();
  bus.set_receiver(receiver_function);
  bus.set_communication_mode(PJON_HALF_DUPLEX);
};

void loop() {
  count++;
  count %= 127;
  if(millis() - time > 10) {
  time = millis();
  transmitter_function();
}
  bus.receive();
  bus.update();
};`

Thank you in advance for your help :-)

gioblu, 2020-03-18:
Ciao @sacreyoubeurt I suspect the problem is `Serial.print`. Try to delete the prints and see if communication gets better. I have experienced that Serial prints can disrupt performance. If you really need the prints, try to call `flush` when finished to print and call `bus.receive()` passing a duration: `bus.receive(1000);` to receive for a millisecond.

gioblu, 2020-03-18:
@sacreyoubeurt I also see here `if(millis() - time > 10) {` 
Are you transmitting cyclically from each node every 10 milliseconds?
That's a tempest of packets I suspect you are saturating the band, specially if running MODE 1

If you need to send a binary state for buttons/switches using a single bit instead of an entire byte.
That would reduce almost by half the length of your packets.

sacreyoubeurt, 2020-03-18:
Hi @gioblu , thank you for your help.
I can't delete Serial.print because there are at the moment my only way to see if communication is okay.
I tried Serial.flush() but it didn't change anything.

However, what you said about bandwidth is interesting for me.
Currently, I'm only using 1 peripheral board and 1 central board : 
- 1 payload of 9 bytes transmitted from periph to central
- 1 payload of 9 bytes transmitted in return  from central to periph 
If I transmit cyclically every 100 ms (and not 10 you're right), I need 9*10*2 = 180 B/s
I did a communication speed test that gave me about 1500 B/s.
So I think it's not the issue.

The question about bandwidth is that in my full project, I will need :
- 1 payload of 9 bytes transmitted from each periph to central.
What you said about buttons is true but this buttons can be replaced by potentiometers so I need to keep at least 1 byte to code value of each one of them.
- 1 payload of 42 bytes transmitted in return from central to each periph
If I transmit cyclically every 100 ms, I will need 9*10 + 42*10 = 510 B/s for each peripheral/central half duplex connection.
So I could only use 2 peripheral board, am i right ?
Should I instead try to send a packet only if a value is changed, packet containing only this value to reduce bandwidth need ?

gioblu, 2020-03-18:
Ciao @sacreyoubeurt for sure one way to reduce the bandwidth use is to transmit only when the value changes. If the module can advertise his own type, the master could know how the data received is formatted. 

Consider that if you use `send_packet` or `send_packet_blocking` you don't need to call `update`. You need to call `update` if you use the `send` or `send_repeatedly` functions.

gioblu, 2020-03-18:
@sacreyoubeurt one thing, once you got it working stable try faster communication modes to have higher bandwidth.

gioblu, 2020-03-18:
@sacreyoubeurt I suggest you to toggle the state of a LED instead of the prints if you are probing communication performance, try also to call `receive` in the slaves passing a longer duration

sacreyoubeurt, 2020-03-22:
Hello there,
I made some improvements in the data transmission frequency but still the half duplex communication between peripheral board and central board is not functional.

When a sensor value changes on peripheral board, it is sent to the central board.
This central affects assigns this value to a MIDI parameter (NoteOn or CC) depending on the type, the Id ans the sensor number of the peripheral. It communicates these MIDI data to the computer as a MIDI class compliant device (it is a flashed Arduino Uno).
This path : peripheral>central>computer is functional.

Then, when computer send MIDI feedback, the central board read and interpret values to find back the Id and sensor number where to send this value (which will be important for motorized faders for example). It works correctly.
But then the transmission of this value by the central or the reception by the peripheral doesn't work and I can't figure it out.
Do you have any idea of what could be the issue ?

Peripheral code : 
`
#define pinVal0 A0
#define pinVal1 A1
#define pinVal2 A2
#define pinVal3 A3
#define pinVal4 2
#define pinVal5 3
#define pinVal6 4
#define pinVal7 5

#define nbrIN 7

#define pinPJON 12
#define PJON_INCLUDE_SWBB
#include <PJON.h>

const int Id = 0; // This peripheral unit bus id
int IdM = 127; // Central bus id
const char Type = 'A'; // This peripheral unit type A: buttons, B: potentiometers, C: encoders, D: faders, E: motorized faders
char TypeM; // Central unit type
int i=0;
uint8_t val[nbrIN];
uint8_t oldval[nbrIN];
int FeedSensNum;
int FeedVal;

PJON<SoftwareBitBang> bus(Id);

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */

  if((char)payload[0] == 'M') {
    IdM = (packet_info.sender_id) ;
    FeedSensNum = ((uint8_t)(payload[1])) ;
    FeedVal = ((uint8_t)(payload[2])) ;
    
    Serial.print(FeedSensNum);
    Serial.println(FeedVal);
    }
}

void transmitter_function() {
 for (int i=0; i<(nbrIN); i++){
  if(oldval[i]!=val[i]){
  uint8_t packet[3] = {
  (uint8_t)(Type),
  (uint8_t)(i),
  (uint8_t)(val[i])
  };
  bus.send_packet(IdM, packet, 3);
  break;
  }
 }
}

void setup() {
  pinMode(pinVal4, INPUT_PULLUP);
[... 133 lines cut ...]
  if (MIDI.read())                // Is there a MIDI message incoming ?
    {
        switch(MIDI.getType())      // Get the type of the message we caught
        {                
                case midi::NoteOn:      // Midi Notes
                MIDIinNote = MIDI.getData1();
                MIDIinVal = MIDI.getData2();
                FeedId = MIDIinNote/8;
                FeedSensNum = MIDIinNote-MIDIinNote/8*8;
                transmitter_note_function();
                break;

                case midi::ControlChange:     // Midi Control Change 
                MIDIinCC = MIDI.getData1();
                MIDIinVal = MIDI.getData2();
                FeedId = MIDIinCC/8+16;
                FeedSensNum = MIDIinCC-MIDIinCC/8*8;
                transmitter_CC_function();
                break;
                }
        }
}
`

Thank you in advance for your help.

gioblu, 2020-03-31:
Ciao @sacreyoubeurt sorry for the late response.
Have you added a pull-down resistor as suggested in the README? https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang

`send_packet` attempts the transmission once, if the channel is busy or interference is present (possibly because of the missing pull-down) packet sending could be aborted.
Try to use `send_packet_blocking` instead of `send_packet`. 

Let me know how it goes :)

gioblu, 2025-11-08:
Closing this for inactivity :)

closed 2025-11-08 by gioblu