gioblu/PJON #338: Possibly unnecessary packet overhead

denizetkar, 2020-03-18:
@gioblu Thank you for this great library, it is written with such a readability that I love reading through it. With that being said, I might be onto something which can be improved here. Although note that I'm not a communication expert like you. However I couldn't help but notice the rather large overhead of a packet from its definition:
```
struct PJON_Packet {
  uint8_t  attempts;
  uint8_t  content[PJON_PACKET_MAX_LENGTH];
  uint16_t length;
  uint32_t registration;
  uint16_t state;
  uint32_t timing;
};
```
The first thing that caught my eyes was `uint16_t state;` . Does a packet really need 16 bits to describe its state, could it not be 8 bits to cut down on some of that overhead? In fact I found that this state variable takes on only the following PRE-defined values:
```
#define PJON_ACK 6
#define PJON_TO_BE_SENT 74
#define PJON_FAIL 65535
#define PJON_BUSY 666
```
Is it a better idea to define a state as uint8_t and change the definitions given above accordingly? If not then I would like to know why and hopefully have a better understanding of how your library works.

gioblu, 2020-03-19:
Ciao @denizetkar thank you for your compliments and your suggestion, you are right, absolutely. 
I have waited to release version 12.1 before answering you, because this are breaking changes that we could include in the next release (13.0). This is clutter sedimented in older versions.

- `receive_frame` could be modified to return 0 in case of failure or the length of the frame
- `receive_response` function could return 0 in case of failure and so have type `uint8_t`
- `PJON_BUSY` and `PJON_FAIL` should be <256
- `PJON_Packet`'s `state` could become `uint8_t` as you correctly suggested
- `compose_packet` could return 0 instead of `PJON_FAIL` and so have type `uint8_t` too
- `send_packet` functions could become type `uint8_t`
- `receive` function could become type `uint8_t`

I don't see any other places affected, do you agree?
Thank you very much again for your valuable contribution.

gioblu, 2020-03-19:
@denizetkar to be clear, `state` is never transmitted, so I am not sure if I would define this as overhead in strict terms, although for sure this change could reduce the amount of used progmem and ram.

denizetkar, 2020-03-19:
> @denizetkar to be clear, `state` is never transmitted, so I am not sure if I would define this as overhead in strict terms, although for sure this change could reduce the amount of used progmem and ram.

Yes you are right! it is indeed not sent as part of the content. Still we could always try to reduce flash and ram usage :)

> Ciao @denizetkar thank you for your compliments and your suggestion, you are right, absolutely.
> I have waited to release version 12.1 before answering you, because those are breaking changes that we could include in the next release (13.0). This is clutter sedimented in older versions.
> 
> * `receive_frame` could be modified to return 0 in case of failure or the length of the frame
> * `receive_response` function could return 0 in case of failure and so have type `uint8_t`
> * `PJON_BUSY` and `PJON_FAIL` should be <256
> * `PJON_Packet`'s `state` could become `uint8_t` as you correctly suggested
> * `compose_packet` could return 0 instead of `PJON_FAIL` and so have type `uint8_t` too
> * `send_packet` functions could become type `uint8_t`
> * `receive` function could become type `uint8_t`
> 
> I don't see any other places affected, do you agree?
> Thank you very much again for your valuable contribution.

As far as I can understand, the `state` variable of a `struct PJON_Packet` can only take 4 different values: `PJON_ACK`, `PJON_TO_BE_SENT`, `PJON_BUSY` and `PJON_FAIL`. So for instance, we instead define them like this:
```
#define PJON_ACK 6
#define PJON_TO_BE_SENT 7
#define PJON_BUSY 111
#define PJON_FAIL 255
```
And define the struct as:
```
struct PJON_Packet {
  uint8_t  attempts;
  uint8_t  content[PJON_PACKET_MAX_LENGTH];
  uint16_t length;
  uint32_t registration;
  uint8_t state;
  uint32_t timing;
};
```

gioblu, 2020-03-19:
> Still we could always try to reduce flash and ram usage :)

It is always a huge pleasure to shave away even few bytes of ram or flash :)

Setting `PJON_FAIL` to 255 would create a problem, `receive_frame` could receive a frame 255 bytes long (the maximum frame length is 65534 bytes), for this reason I was suggesting to use 0 instead (`receive_frame` will never receive a frame 0 bytes long), same thing for `compose_packet`. Then if `PJON_FAIL` and `PJON_BUSY` can be expressed with 8 bits `receive`, `receive_response` and `send_packet` could consequently become of type `uint8_t`. 

What do you think?
I think the footprint reduction will not be negligible!

denizetkar, 2020-03-20:
First of all, excuse me for this late reply. I felt the need to thoroughly analyse parts of the library that will be affected by the suggested changes to `PJON_FAIL` and `PJON_BUSY`. Here are the complete list of functions in `PJON.h` that will be either directly or indirectly affected:
```
PJON::dispatch(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *packet,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint16_t packet_index = PJON_FAIL
    )
PJON::receive()
PJON::receive(uint32_t duration)
PJON::handle_asynchronous_acknowledgment(PJON_Packet_Info packet_info)
PJON::reply(
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send_from_id(
      uint8_t tx_id,
      const uint8_t *tx_bus_id,
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send_repeatedly(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send_repeatedly(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send_packet_blocking(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint32_t timeout = 3500000
    )
PJON::send_packet_blocking(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint32_t timeout = 3000000
    )
PJON::send_packet(const uint8_t *payload, uint16_t length)
PJON::send_packet(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::send_packet(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    )
PJON::update()
```
As I was saying, I was analysing whether these functions would behave the same way as before. Although I'm not finished with all of them yet, I found that `PJON::dispatch` will behave differently and here is why:

- Let's define `#define PJON_FAIL 0`,
- Then upon calling `PJON::dispatch` without providing `uint16_t packet_index` argument (defaults to `PJON_FAIL`), the new `PJON::dispatch` function will behave (compared to the current version) as if `packet_index == 0` and it will compose a new packet and insert it into `PJON_Packet PJON::packets`. INSTEAD, it should have always skipped composing a packet/inserting it and it should always execute:
```
      _error(PJON_PACKETS_BUFFER_FULL, PJON_MAX_PACKETS, _custom_pointer);
      return PJON_FAIL;
```
So we will first need to come up with a way to preserve the behavior of `PJON::dispatch` function with respect to the suggested changes. I will be looking into the behavioral changes in the other affected functions as well, as soon as I can.

denizetkar, 2020-03-20:
Here is a more comprehensive analysis of the affected functions in `PJON.h`. I found by manual inspection that there are 3 main types of return values for these functions and they are namely `FRAME_LENGTH`, `PACKET_INDEX` and `PJON_OPERATION_STATE`. As their names imply they respectively relate to "length of the processed frame", "index or count of packet(s)" and "state of the operation (whatever that may be)". Then I categorized each affected function according to their return type as follows:
```
PJON::compose_packet(
      const uint8_t rx_id,
      const uint8_t *rx_bus_id,
      uint8_t *destination,
      const void *source,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= FRAME_LENGTH

PJON::dispatch(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *packet,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint16_t packet_index = PJON_FAIL
    ) ::= PACKET_INDEX

PJON::receive() ::= PJON_OPERATION_STATE

PJON::receive(uint32_t duration) ::= PJON_OPERATION_STATE

PJON::reply(
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send_from_id(
      uint8_t tx_id,
      const uint8_t *tx_bus_id,
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send_repeatedly(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send_repeatedly(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint32_t timing,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PACKET_INDEX

PJON::send_packet(const uint8_t *payload, uint16_t length) ::= PJON_OPERATION_STATE

PJON::send_packet(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PJON_OPERATION_STATE

PJON::send_packet(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST
    ) ::= PJON_OPERATION_STATE

PJON::send_packet_blocking(
      uint8_t rx_id,
      const uint8_t *rx_bus_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint32_t timeout = 3500000
    ) ::= PJON_OPERATION_STATE

PJON::send_packet_blocking(
      uint8_t rx_id,
      const void *payload,
      uint16_t length,
      uint8_t  header = PJON_NO_HEADER,
      uint16_t packet_id = 0,
      uint16_t rx_port = PJON_BROADCAST,
      uint32_t timeout = 3000000
    ) ::= PJON_OPERATION_STATE

PJON::update() ::= PACKET_INDEX
```
What we can do with this information is that, we can freely change the definitions of `PJON_FAIL` and `PJON_BUSY` then use these only in the functions that has a return type of `PJON_OPERATION_STATE`. And for the other functions (with return types of `PACKET_INDEX` and `FRAME_LENGTH`) we can define another set of MACROs that are of type `uint16_t` so that for example `FRAME_LENGTH` is still of type `uint16_t` (because the maximum frame length is 65534 bytes). We can also preserve the behavior of `PJON::dispatch` because it has a return type of `PACKET_INDEX` which I suggest that it should have different MACROs (it should not use `PJON_FAIL`, instead it should use its own MACROs for `PACKET_INDEX` type).

Do you think this is a reasonable solution to define another set of macros for functions with both return types `PACKET_INDEX` and `FRAME_LENGTH`?

gioblu, 2020-03-20:
Ciao @denizetkar thank you for your valuable analysis. You are right, yes I have missed `dispatch` issue when using 0 for `PJON_FAIL`. Although, I think your suggestion has sense, and I find a nice idea to use macros for the types you have correctly categorised. Would you propose the changes to record your contribution or you prefer me to work on the implementation as you suggested?

Thank you very much again for your contribution.

gioblu, 2020-06-14:
Ciao @denizetkar I have made some tests, the few bytes spared don't value a breaking change specially if compounded with many others that are coming out in the next release, I will re-evaluate this in the next iteration. Thank you for your support.

closed 2020-06-14 by gioblu