gioblu/PJON #339: question: implemented on top of radiohead?

jhoogeboom, 2020-03-23:
Did you consider, or has anyone looked into implementing this protocol on top of RF libraries like RadioHead? This is a software library for interfacing with HopeRF and a bunch of other popular transceivers.
https://www.airspayce.com/mikem/arduino/RadioHead/

This seems to be the interface to implement on the PJON side:
https://github.com/gioblu/PJON/tree/master/src/strategies

gioblu, 2020-03-30:
Ciao @jhoogeboom sorry for the late reply, thank you for your suggestion.
For sure a RadioHead strategy would be a good thing, I will look into RadioHead as soon as possible. If you have the hardware and the will to try to implement it feel free to propose a pull-request.

jhoogeboom, 2020-04-01:
Hey no worries. I am interested and I should have some hardware somewhere, although it might not be on a very quick timeframe. Should be relatively easy, although RadioHead has a lot of routing functionality that will go unused.