gioblu/PJON #341: ESP8266 & Nano in PJON<SoftwareBitBang> incompatibility: possible cause found?

rapnagem, 2020-03-29:
Hi there, 
I have been trying for a while to make a "NodeMCU 1.0 (ESP-12E Module)" communicate to an "Arduino Nano (ATmega328P - Old Bootloader)" using I2C (bi-directional communication).
I tried a few PJON examples with the following hardware pins configuration, and both serial ports connected to the computer USB ports for software upload:

```
ESP8266(Vin) --> ArduinoNano(Vin)
ESP8266(Gnd) --> ArduinoNano(Gnd)
ESP8266(D1) --> ArduinoNano(12)
```
When running

[ESP8266>Local>SoftwareBitBang>SpeedTest>ESPTransmitter](https://github.com/gioblu/PJON/blob/master/examples/ESP8266/Local/SoftwareBitBang/SpeedTest/ESPTransmitter/ESPTransmitter.ino) in the ESP8266 board &
[ESP8266>Local>SoftwareBitBang>SpeedTest>Receiver](https://github.com/gioblu/PJON/blob/master/examples/ESP8266/Local/SoftwareBitBang/SpeedTest/Receiver/Receiver.ino) in the Arduino Nano board 
they seem to work almost ok, with the following result obtained in receiver Arduino Serial port (this is a general result). Apart from the "Fail (no data found): 1909" info, it seems almost ok to me. 
```
 --------------------- 
Starting 1 second communication speed test...
Packet Overhead: 10B - Total: 540B
Bandwidth: 1620.00B/s
Data throughput: 1080.00B/s
Packets sent: 54.00
Mistakes (error found with CRC): 0.00
Fail (no data found): 1909
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
 --------------------- 
```
I then run
[ARDUINO>Local>SoftwareBitBang>SendArbitraryDataType>Transmitter](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/SendArbitraryDataType/Transmitter/Transmitter.ino) in the ESP8266 board &
[ARDUINO>Local>SoftwareBitBang>SendArbitraryDataType>Receiver](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/SendArbitraryDataType/Receiver/Receiver.ino) in the Arduino Nano board

After 2 adaptations in the transmitter code 
```ruby
unsigned long mytime;                      // instead of unsigned long time;       
bus.strategy.set_pin(D1);                  // instead of bus.strategy.set_pin(12);    
```

the ArduinoIDE was able to compile both codes, but NO information was printed into the receiver Serial port (Arduino Nano). So, I made a third adaptation in the transmitter code (ESP8266)
```ruby
bus.set_communication_mode(PJON_SIMPLEX);         // this was included;  
```
I could finally read the received data in the Arduino Nano serial port (a general result is shown below):
```
Receiver device id: 44 | Transmitter device id: 45
Record voltage: 19mV | Record id: 243990528

Receiver device id: 44 | Transmitter device id: 45
Record voltage: 19mV | Record id: 244121600

Receiver device id: 44 | Transmitter device id: 45
Record voltage: 29mV | Record id: 244318208
...
```

Surprisingly, while the "voltage" values appear to be correct, the "id" values are absolutetly wrong; they are supposed to start from 1 and increase sequentially; according to the code. **Said that, I believe there is an inconsistency between the sent/received data protocols for these boards.**

For curiosity, I changed the order in the way "struct voltage_record" object is created, changing it in both the transmitter and receiver codes, as following:
```ruby
struct voltage_record {
 uint32_t id; 
 uint16_t voltage;
}
``` 
The output then became 
``` 
Receiver device id: 44 | Transmitter device id: 45
Record voltage: 19mV | Record id: 1

Receiver device id: 44 | Transmitter device id: 45
Record voltage: 14mV | Record id: 3

Receiver device id: 44 | Transmitter device id: 45
Record voltage: 14mV | Record id: 5

...
``` 

Surprisingly, the id´s are now only odd numbers, while they were suppossed to be 1,2,3... and so on.

I hope this may help to solve the problem and further improve this great PJON library.

I have not tested to send any data back to the ESP8266 so far, but I will do this soon.

rapnagem, 2020-03-29:
Hi, I came back to this problem today and the following changes basically solved the problem
```
struct voltage_record {
 uint16_t id;                            //we are now limited to an id value of 65535.
 uint16_t voltage;
}

bus.update();                           //to be included at the end of the void loop call
```
meanwhile uint32_t and other types are not correctly handled.

gioblu, 2020-03-30:
Thank you very much @rapnagem, I will patch the examples as you well suggested. Consider also that, the `Serial.print` calls may disrupt SoftwareBitBang communication, so that may be the cause of receiving only odd ids, each exchange a packet may go lost because of Serial print interference.

gioblu, 2020-03-30:
If that happens `Serial.flush` may help (avoinding parallel or simultaneous transmission of Serial and SoftwareBitBang data).

closed 2020-03-30 by gioblu

rapnagem, 2020-03-30:
Hi Giovanni
First of all, I would like to wish to you and all my latin colleagues, all the best with the actual situation in Italy. I have been there once, and I do hope you all can get over this as soon as possible.
Secondly, thank you very much for your kind replies. It was really good to hear from you and to see that you are doing a great library! Congratulations.
Yes. You are absolutely right!

I placed the Serial.flush at end of every cluster of Serial.print commands (including at the void setup and/or void loop and/or void function) and also changed r.id to uint32_t and it worked.
Aparently the examples are also missing such commands as you said.
My best regards
Ronaldo.

---------------------------------------------------------
Prof. Dr. Ronaldo Alves Pinto Nagem
Universidade Federal de Minas Gerais
Instituto de Ciências Biológicas
Departamento de Bioquímica e Imunologia
Av. Antônio Carlos, 6627 - Caixa Postal 486
Bairro Pampulha - CEP: 31270-901
Belo Horizonte, MG - Brasil
Tel: +55 31 3409-2626
Fax: +55 31 3409-2614
E-mail: nagem@icb.ufmg.br
E-mail: rapnagem@yahoo.com.br

gioblu, 2020-03-30:
Thank you for your warm words @rapnagem I hope all will get resolved worldwide as soon as possible. Will soon follow a referenced commit with the patch we have identified.

Thank you again.
Stay safe.

rapnagem, 2020-03-30:
Hi @gioblu I removed my previous comment as it had a lot of personal information on it... hahahaha (first time using the GitHub). But the message was sent to you! Keep yourself safe too!
Placing the ```Serial.flush()``` at the end of every cluster of ```Serial.print()``` commands (included in the ```void loop {}```, ```void setup{}``` and ```void functions{}``` ) solved the issue. Congrats for such great library.

gioblu, 2020-03-30:
No probs @rapnagem :)