gioblu/PJON #342: ThroughLora strategy for ESP32?

elimisteve, 2020-03-29:
Some friends of mine are working on a couple really awesome mesh networking projects, namely https://disaster.radio/ ("a disaster-resilient communications network powered by the sun") and https://peoplesopen.net/ ("People's Open is a community-owned and -operated wireless network in Oakland, California.Open source, peer-to-peer internet").

They are using the ESP32 and I told them about PJON.

As per [the compatibility table](https://github.com/gioblu/PJON/wiki#compatibility), I see that the TR/ThroughLora strategy does not yet work with the ESP32.

Are there plans to add it?

Thanks!

gioblu, 2020-03-30:
Ciao @elimisteve thank you for your support and for spreading the word. Yes, the support for ESP32 is for sure necessary, I will try to work it out as soon as possible.

paidforby, 2020-03-30:
For what it's worth, the example [ThroughLoRa HalfDuplex code](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/ThroughLoRa/HalfDuplex/Device1/Device1.ino) compiles and doesn't crash when flashed to the ESP32 LILYGO TTGO V2.1 dev board. It doesn't do much else, so it's hard to tell if the LoRa transceiver is being initialized, but at least that's a start.

gioblu, 2020-03-30:
Thank you @paidforby I will give it a try with a couple of Heltec WiFi LoRa and report back here.
Compliments for the work done on disaster-radio, it is really a cool project. I suspect AnalogSampling: https://www.pjon.org/AnalogSampling.php may be a really cool addition to its support, it could be possible to connect nodes that have line of sight wirelessly using lasers diodes and just LEDs or dedicated photodiodes as receivers.

paidforby, 2020-03-30:
Sure, PJON definitely looks like a valuable protocol that we would want to support. We'll discuss the progress on support in our repo here, https://github.com/sudomesh/disaster-radio/issues/63.