gioblu/PJON #343: simulating one esp32 master - two d1-mini slaves cable connection

fotosettore, 2020-04-03:
ciao
first : my compliments for PJON library !!
It seems solve my problem about this scenario:
I need to connect throught cable (no wifi - no BT) one ESP32 as master and two D1-MINI as slaves.
I saw a lot of example inside here but i'm really confused. 
May you help me to indicate which is the .ino that should be the near example to solve my problem ?
many thanks

gioblu, 2020-04-03:
Ciao @fotosettore, thank you for your compliments, if you need a single cable connection you are looking for [SoftwareBitBang](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) the examples are [here](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang) just change the pin used in the examples to reflect your setup.

fotosettore, 2020-04-03:
ciao gioblu
i found two errors in TRASMITTER.INO  in 
PJON-master\examples\ARDUINO\Local\SoftwareBitBang\SendArbitraryDataType:

1. SOLVED
line 8 : 
`unsigned long time;`
must to be changed in (example) 
`unsigned long Mytime;`
because time is now a reserved reserved word
obviously the same for line 31

2. NOT SOLVED
during compilation, it stops in line 15 with following error :
`"previous definition of 'struct_voltage_record'`

Please, may you help me for not solved error ?
many thanks
peppe

gioblu, 2020-04-03:
Ciao @fotosettore I have fixed the errors you have reported, thank you very much.
I have tried to compile the examples using the Arduino IDE for various ESP boards, such as NodeMCU and Wemos D1 mini, I cannot preproduce the redefinition error, are you sure you haven't duplicated the definition in your code?

fotosettore, 2020-04-03:
ciao gioblu !
**ERROR 2 SOLVED**
sorry sorry sorry
usually i make a copy of original .ino.and I put it in a temporary folder using it for my test.
i forgot to move it and - of course - the second file was as  to belong the same project, with duplicate definition.
A beer is ready to drink for free in virtual pub of gioblu-github !

So one master  <-> one slave configuration works fine !

In few time I'll test the configuration about "one-master <--> more slave and i will keep you and all users informed.

Stammi bene !
peppe

gioblu, 2020-04-03:
Ciao @fotosettore peppe, grazie mille del tuo report.
Stammi bene anche tu, speriamo che sta storia finisca presto.
Un abbraccio.

closed 2020-04-03 by gioblu