gioblu/PJON #344: PJON<SoftwareBitBang> bus(xx) inside void setup

fotosettore, 2020-04-12:
hi folks!
my question is simply :
is there a way to put slave number inside void setup ?
at moment, the number is out of setup,so, if i need to change on the fly, it is not possibile

i need something similar : 

`      pinMode(SlavePin, INPUT);
      if (digitalRead(SlavePin) == HIGH)
      {
        PJON<SoftwareBitBang> bus(50);
      }
      
      if (digitalRead(SlavePin) == LOW)
      {
        PJON<SoftwareBitBang> bus(51);
      }  `

So if i put SlavePin to ground, slave number will be 51, if i put it to +3v3, it will be 50

At moment the number is gave inside compilation

ciao
peppe

gioblu, 2020-04-12:
Ciao @fotosettore, buona Pasqua, you can instantiate an object without passing an id, for example:
```cpp
PJON<SoftwareBitBang> bus;
```
The device id is by default `PJON_NOT_ASSIGNED`, until you use `set_id`, for example:
```cpp
// In the setup or loop
bus.set_id(51); // set device id of the instance to 51
```
Device id 0 or `PJON_BROADCAST` is used as broadcast address, so avoid it.
I will soon answer to your email :)

closed 2020-04-12 by gioblu

fotosettore, 2020-04-12:
hi gioblu
i'm really happy that you are ok ... buona pasqua anche a te !!!
i tried and, naturally, all works fine !
i will use this way to choose the right id number on many devices with only a compile job.
i tried on esp32 (all types) and I have never had any errors.
my last test was on esp32 mini 
[https://images-na.ssl-images-amazon.com/images/I/61o9SHnESdL._AC_SL1000_.jpg](url)

p.s. 
>>>I will soon answer to your email :)
grazie !!! attendo con ansia per poter andare avanti nella mia progettazione. Al momento sono fermo in attesa di capire se è possibile quanto ti ho richiesto, altrimenti debbo prendere altre decisioni in merito e non vorrei farlo :-(