gioblu/PJON #346: Support for Attiny402

kopierreko, 2020-04-26:
Is it possible to support Attiny402 ?

gioblu, 2020-04-27:
CIao @kopierreko it may work although you would not have much memory left for the program.

kopierreko, 2020-04-27:
It's what I encountered when I tried to modify the sources... I was searching for a solution to make a one wire slave with attiny402 and I found your projet which look really nice but there is this problem of memory now ^^
Thanks 😊

gioblu, 2020-04-27:
Ciao @kopierreko the smaller MCU PJON supports is ATtiny85, that is already a feat, I am not sure we will ever be able to support the 4k family. You may have luck using the SoftwareBitBang strategy alone, and design for it, a custom, simpler, packet format.

closed 2020-04-27 by gioblu