gioblu/PJON #347: traffic light on SoftwareBitBang bus.reply

fotosettore, 2020-04-28:
Ciao Giovanni,
I succesfully connected 3 (and more) slaves using SoftwareBitBang . The scenario is that may be possible that all togheter or two of them send a **bus.reply** in the same time.
Master cannot manage the answer because i saw that go in trouble for it.
Is there a way to have a "traffic light" for that ? In other words something that creates a RX queue so no data will be lost ?
Peppe
p.s. spero tu stia bene.

gioblu, 2020-04-29:
Ciao @fotosettore all fine here, I hope all is ok there too. SoftwareBitBang operates using a variation of CSMA, for this reason collision may occur in some cases. For example, if from a master you send a packet that means "send me back your state" many devices could answer back mostly at the same time, potentially colliding. So if you need to ask the state from a master to the slave it is more reliable to implement the request-response scheme, in which master asks to each slave for its state sequentially. Let me know how it goes!

gioblu, 2020-04-29:
@fotosettore cool :) feel free to call me if you need, sorry if I disappeared but I had to focus on the new changes

closed 2020-04-29 by gioblu