gioblu/PJON #348: AT commands on HC-11

rwlakin, 2020-04-29:
No AT response using TeraTerm virtual terminal.  Works fine on all bluetooth modules. Trying several HC-11 modules.  Pin 5 to gnd.

gioblu, 2020-04-29:
Ciao @rwlakin I have recently bought Reyax LoRa modules to implement AT commands support. For now PJON using ThroughSerial or ThroughSerialAsync can work with HC-12 and other "transparent" modules. 

>  Works fine on all bluetooth modules
I have never tried bluetooth modules, would you tell me the models so I can list them in the documentation? Have you used ThroughSerial or ThroughSerialAsync?

Thank you for your support.

rwlakin, 2020-04-29:
Problem solved.  I stop using TeraTerm and switch to CoolTerm and it works
fine.  Haven't tried the ESP8655 module yet.

closed 2020-04-29 by gioblu